WebODM/WebODM #1087: Indication of completed calculation

APOS80, 2021-11-06:
### What's your browser and operating system?
Chrome Win 10

### What is the problem?

Hard to se what steps is done.

### What should be the expected behavior?

I would like to se a dot or colouring to indicate if a step is calculated/done.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

pierotofy, 2021-11-06:
Hey @APOS80 we'd welcome a PR to improve this.

APOS80, 2021-11-06:
> Hey @APOS80 ✋ we'd welcome a PR to improve this.

Don’t know what I can do, I know some python though. Wouldn’t know where to begin actually.

pierotofy, 2021-11-08:
The frontend is Javascript/React. It could be a fun learning experience!

https://reactjs.org/

APOS80, 2021-11-08:
> The frontend is Javascript/React. It could be a fun learning experience!
> 
> https://reactjs.org/

Got a jobb and a family, i hardly have time to breath

pierotofy, 2021-11-12:
Closing, I'd welcome a PR that implements this. :pray:

closed 2021-11-12 by pierotofy