WebODM/WebODM #1088: Error when computing

APOS80, 2021-11-06:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome, Windows 10

### What is the problem?

Error when processing

### What should be the expected behavior?

No error

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 3221226505
Traceback (most recent call last):
File "C:\WebODM\resources\app\apps\ODM\stages\odm_app.py", line 94, in execute
self.first_stage.run()
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 341, in run
self.next_stage.run(outputs)
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 341, in run
self.next_stage.run(outputs)
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 341, in run
self.next_stage.run(outputs)
[Previous line repeated 1 more time]
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 322, in run
self.process(self.args, outputs)
File "C:\WebODM\resources\app\apps\ODM\stages\openmvs.py", line 79, in process
system.run('%s "%s" %s' % (context.omvs_densify_path,
File "C:\WebODM\resources\app\apps\ODM\opendm\system.py", line 106, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 3221226505

===== Done, human-readable information to follow... =====

[ERROR]   The program exited with a strange error code. Please report it at https://community.opendronemap.org

pierotofy, 2021-11-09:
Does this error happen all of the times or just with some images/settings?

APOS80, 2021-11-10:
> Does this error happen all of the times or just with some images/settings?

Happend when processing 558 images from a Mavic 2 Pro, Ultra high setting, ORB. Possible lack of memmory but I dont know.

pierotofy, 2021-11-10:
Ok so not all of the times.

pierotofy, 2022-05-20:
Feel free to re-open with a link to your dataset? Thanks.

closed 2022-05-20 by pierotofy