WebODM/WebODM #1092: Measurements not completing, stuck on "computing"

jaharvey86, 2021-11-10:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 95 on Windows 10

### What is the problem?

I am viewing the 2D Orthophoto in WebODM 1.9.11 and attempting to use the measurement tool to measure the volume of an area. When I click "finish measurement" it just says "computing..." with the cog constantly turning. The volume is never calculated. I have tried several different areas to be measured, even the smallest areas do the same thing.

### What should be the expected behavior?

After a few seconds to a minute, "computing" should disappear and it presents the volume

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Run a set of approx 200 images through for DSM and DTM.
Click view Map
Click "measure volume, area and length"
Click "Create a new measurement"
Select three areas of the map to measure to create a small triangle
Click "Finish measurement"
Observe fault.

[image]

pierotofy, 2021-11-10:
Thanks for the report @jaharvey86 :pray: this should be fixed, please update and if it isn't re-open the issue?

jaharvey86, 2021-11-11:
Hi, thanks for getting back to me. It's not fixed for me, I am a user of WebODM via the Window Installer. Do I need to do anything to update it and implement your fix from 1093?

pierotofy, 2021-11-11:
When you launch the application, does it read: `version 1.9.11 build 38`? If not, you need to Update via `Tools -- Check for Updates`.

jaharvey86, 2021-11-11:
I completely uninstalled build 37 and re-installed with build 38, ran a new set of images and I still get the same issue.

pierotofy, 2021-11-12:
Could you attach the `app.log` file after you perform a volume measurement? (From `Tools` --> `View Logs`).

jaharvey86, 2021-11-12:
[app.log](https://github.com/OpenDroneMap/WebODM/files/7527022/app.log)

App log is uploaded. I ran a new set through it too but experienced the same issue. Thank you.

pierotofy, 2021-11-12:
Thanks for the log! I've pushed another update (build 39), could you see if this one fixes it? :pray:

jaharvey86, 2021-11-15:
hi, this build still doesn't resolve it. I've attached another log if that helps?

I am unable to reopen this issue (it says it's closed), should I open a new one?

[app.log](https://github.com/OpenDroneMap/WebODM/files/7541114/app.log)

pierotofy, 2021-11-15:
Thanks @jaharvey86 , this issue has been tricky to fix. No need to open a new issue.

I've just published a new build, which should hopefully fix this for good (update and retry once more?) :pray:

pierotofy, 2021-11-16:
Build `42` is the latest.

jaharvey86, 2021-11-17:
I can confirm that this issue is resolved in Build 43. Thank you very much for all your help.

PR #1093 (2021-11-10): GRASS use tmp-location, translation fix
Closes #1092﻿
closed 2021-11-17 by jaharvey86