WebODM/WebODM #1096: Possible Bug: geo.txt not being auto-detected nor loaded

Saijin-Naib, 2021-11-16:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
NA

### What is the problem?
A properly named/structured geo.txt that is uploaded into WebODM alongside the referenced images does not appear to be used by WebODM.

### What should be the expected behavior?
The geo.txt should be auto-detected and used.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Community Post (with sample data/geo.txt):  
https://community.opendronemap.org/t/cannot-pass-image-geolocation-file-geo-txt-into-project/6368/20?u=saijin_naib

pierotofy, 2021-11-16:
I found a separate issue, the file is detected, but is interpreted as a GCP if image resize is turned on, which obliterates its contents.

gastgit, 2021-12-16:
hi.. Is this resolved?

pierotofy, 2021-12-16:
I don't think it is.

gastgit, 2021-12-16:
> I don't think it is.

Tnx @pierotofy 
Was it only pertinent to WebODM as far as you understood or a more general issue?

pierotofy, 2022-02-28:
Should be fixed now!

closed 2022-02-28 by pierotofy

gastgit, 2022-02-28:
> Should be fixed now!

Tnx!