WebODM/WebODM #1098: Run it natively

ccooddeerr, 2021-11-23:
Hi, could you update the script version to run it natively? I‘m trying to install it on my ubuntu 18.04 machine without docker.

pierotofy, 2021-11-23:
Hi @ccooddeerr,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-11-24 02:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-11-23:
Hi, we'd welcome a pull request to update the script. We don't officially support non-docker installations. :pray:

closed 2021-11-23 by pierotofy