WebODM/WebODM #1099: Add compression to GeoTIFF export outputs

pierotofy, 2021-11-25:
We currently do not compress outputs when generating GeoTIFFs. This is not ideal.

Saijin-Naib, 2021-11-25:
I think we've got some room to improve, especially with modern GDAL and other dependencies.

We should be able to do band interleaved with predictor 2, and maybe ZSTD+LERC for DEMs and maybe JPEG + transparency band for orthophotos...

https://github.com/OpenDroneMap/ODM/issues/1040
https://kokoalberti.com/articles/geotiff-compression-optimization-guide/
http://blog.cleverelephant.ca/2015/02/geotiff-compression-for-dummies.html

pierotofy, 2021-11-25:
I would love to use the latest and greatest, just not sure if older/proprietary programs will be able to read the results afterwards. Needs to be balanced for interoperability.

Saijin-Naib, 2021-11-25:
DEFLATE + Predictor 2 and ZLEVEL 12 or 9 should be about as good as we can get while keeping things bog-standard then.

closed 2022-02-28 by pierotofy

andrewharvey, 2026-08-19:
https://github.com/linz/elevation/blob/master/docs/research/tiff-compression.md has some analysis