WebODM/WebODM #1100: Hard recovery guide #impovements

markFieldman, 2021-11-25:
Hi, recently I have faced a hard-recovery issue when I had only a media folder, but the webodm_db backup was corrupted. 
I create a hard_recovery script which will recreate a project structure in db from media folder project and task.
Here it is

### Use this script in manage.py shell of webODM for HARD RECOVERY

### CAUTION - ALL OPERATIONS AND HARM FROM THIS SCRIPT - YOUR OWN RESPONSIBILITY !!!
### USE THIS SCRIPT ONLY IF THERE AREN`T ANOTHER WAY !!!

## 1. Copy and paste imports and functions to py shell
## START COPY FIRST PART
from django.contrib.auth.models import User
from app.models import Project, Task, ImageUpload
import os
from django.contrib.gis.gdal import GDALRaster
from django.contrib.gis.gdal import OGRGeometry
from django.contrib.gis.geos import GEOSGeometry
from django.db import connection
from django.utils.translation import gettext_lazy as _, gettext
from nodeodm import status_codes
from app.cogeo import assure_cogeo
import json

def process_task(creating_task):
    images_json = creating_task.assets_path("images.json")
    if os.path.exists(images_json):
        try:
            with open(images_json) as f:
                images = json.load(f)
                creating_task.images_count = len(images)
        except:
            print("Cannot read images count from imported task {}".format(creating_task))
            pass
    extent_fields = [
        (os.path.realpath(creating_task.assets_path("odm_orthophoto", "odm_orthophoto.tif")),
         'orthophoto_extent'),
        (os.path.realpath(creating_task.assets_path("odm_dem", "dsm.tif")),
         'dsm_extent'),
        (os.path.realpath(creating_task.assets_path("odm_dem", "dtm.tif")),
         'dtm_extent'),
    ]
    for raster_path, field in extent_fields:
        if os.path.exists(raster_path):
            # Make sure this is a Cloud Optimized GeoTIFF
            # if not, it will be created
            try:
                assure_cogeo(raster_path)
            except IOError as e:
                print(
                    "Cannot create Cloud Optimized GeoTIFF for %s (%s). This will result in degraded visualization performance." % (
                    raster_path, str(e)))
            # Read extent and SRID
            raster = GDALRaster(raster_path)
            extent = OGRGeometry.from_bbox(raster.extent)
            # Make sure PostGIS supports it
            with connection.cursor() as cursor:
                cursor.execute("SELECT SRID FROM spatial_ref_sys WHERE SRID = %s", [raster.srid])
                if cursor.rowcount == 0:
                    raise Exception()
            # It will be implicitly transformed into the SRID of the model’s field
            # task.field = GEOSGeometry(...)
            setattr(creating_task, field, GEOSGeometry(extent.wkt, srid=raster.srid))
            print("Populated extent field with {} for {}".format(raster_path, creating_task))
            creating_task.update_available_assets_field()
            creating_task.potree_scene = {}
            creating_task.running_progress = 1.0
            creating_task.console_output += gettext("Done!") + "\n"
            creating_task.status = status_codes.COMPLETED
            creating_task.save()

def create_project(project_id, user):
    project = Project()
    project.name = project_id
    project.owner = user
    project.id = int(project_id)
    return project

def reindex_shots(projectID, taskID):
    project_and_task_path = f'project/{projectID}/task/{taskID}'
    try:
        with open(f"/webodm/app/media/{project_and_task_path}/assets/images.json", 'r') as file:
            camera_shots = json.load(file)
            for image_shot in camera_shots:
                ImageUpload.objects.update_or_create(task=Task.objects.get(pk=taskID),
                                                 image=f"{project_and_task_path}/{image_shot['filename']}")
                print(f"Succesfully indexed file {image_shot['filename']}")
    except Exception as e:
        print(e)

## END COPY FIRST PART

## 2. Specify the username which will have acess to the imported projects
user = User.objects.get(username="YOUR NEW CREATED USER HERE")

## 3. This is the main part of script which make the main magic of the project
for project_id in os.listdir("/webodm/app/media/project"):
    if not Project.objects.filter(id=int(project_id)).exists():
        project = create_project(project_id, user)
        project.save()
    else:
        project = Project.objects.get(pk=(project_id))
    for task_id in os.listdir(f"/webodm/app/media/project/{project_id}/task"):
        if not Task.objects.filter(id=task_id).exists():
            task = Task(project=project)
            task.id = task_id
            process_task(task)
            reindex_shots(project_id, task_id)

## 4. Update project ID sequence for new created tasks
with connection.cursor() as cursor:
                cursor.execute("SELECT setval('app_project_id_seq', (SELECT MAX(id) FROM app_project)+1)")

Can we add this guide to the readme as part of the HARD RECOVERY for advanced users?

pierotofy, 2021-11-25:
Hi @markFieldman,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-11-26 04:35) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-11-25:
Thanks @markFieldman :hand: this could be useful indeed; perhaps you can open a pull request with a script to place in `contrib`? https://github.com/OpenDroneMap/WebODM/tree/master/contrib

You can also add a README.md in that folder with instructions and then link it from the main readme.

closed 2021-12-01 by pierotofy