WebODM/WebODM #1101: Processing failed (4294967295)

studiosarchese, 2021-11-28:
### What's your browser and operating system? 

[image]
Edge 96 on Windows 11

### What is the problem?

In fase di processamento immagini restituisce codice errore: Processing failed (4294967295)

### What should be the expected behavior?

Dovrebbe processare le immagini e restituire i dati di elaborazione

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Ho caricato le immagini riprese con un drone Mavic Air 2 pro, le ho importate tramite il tool del software, in nodo elaborazione ho lasciato auto, in opzioni ho lasciato predefinito, ridimensiona immagini ho lasciato no e infine ho lanciato l'elaborazione. Ho provato su due pc è da lo stesso errore.

[image]

pierotofy, 2021-12-02:
@studiosarchese can you please provide a copy of your images?

Karanza, 2022-06-25:
Ciao,ho il tuo stesso problema.Sei riuscito a risolvere?

Saijin-Naib, 2022-06-25:
@Karanza , Are you on the current version (v1.9.14 Build 64)? This should not be happening anymore on the latest releases.

Karanza, 2022-06-25:
Si,ho la versione che dici te.Pero ho notato che se nelle opzioni metto model 3d fallisce,se invece messo personalizzato va a buon fine...non capisco il perché? Il voglio fare modello 3 d,ma come faccio?

Saijin-Naib, 2022-06-25:
Apri una discussione sulla nostra Community poiché il tuo problema è diverso:
https://community.opendronemap.org/invites/DGsLLWf1tr

pierotofy, 2022-07-22:
No input images, cannot reproduce. Closing.

closed 2022-07-22 by pierotofy

Karanza, 2022-10-11:
Si,ho quella versione...ho notato che se nelle opzioni metto modello 3d fallisce e da errore, mentre se metto personalizzato il processo va a buon fine.. Come mai?

Andrea

-------- Messaggio originale --------

Data: 25/06/22 03:54 (GMT+01:00)
A: OpenDroneMap/WebODM ***@***.***>
Cc: Karanza ***@***.***>, Mention ***@***.***>