WebODM/WebODM #1103: Can't upload to OAM from WebODM

cgiovando, 2021-11-30:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Mac Pro Server (?)

### What is the problem?

This was reported directly to the OAM team from a WebODM user, not able to upload to OAM. They provided the screenshots below and the source image. The issue may be caused by the local file OAM uploader which seems to be broken at the moment.

### What should be the expected behavior?

Should be able to upload from WebODM to OAM

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

GeoTIFF mosaic  https://cloud.collectivetech.ca/s/fyfF76yz8EzEcTs

[image]

[image]

pierotofy, 2021-12-01:
I can't seem to reproduce this on the latest version (docker); I was able to upload a dataset fine.

Can you reproduce this directly @cgiovando or provide more information about how WebODM was installed?

torrinjaroswanson, 2021-12-01:
The Firefox browser initiating the upload from OAM and OAM sharing from webobm
```

    Software Type
    Web Browser

    Hardware Type
    Computer

    Browser version (full)
    94.0 (94.0.2 (64-bit))

    Operating System version (full)
    10.15

    Layout Engine
    Gecko

    Gecko version
    20100101

    Configured Language
    English

    Device Pixel Ratio
    2

    Browser GMT Offset
    -06:00

    Navigator Platform
    Intel based Mac

    OS / CPU
    Intel Mac OS X 10.15

    No. of logical CPU cores
    6

    Maximum Touch Points
    0

    WebGL Vendor
    Intel Inc.

    WebGL Renderer
    Intel(R) HD Graphics 400

    ECMA Version
    11 (ES2020)

    User agent:
    Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:94.0) Gecko/20100101 Firefox/94.0
```

Server with WebODM installed:
```
Make/Model
2009 Mac Pro

Kernel:
Linux 5.11.0-40-generic on x86_64

OS:
Ubuntu 20.04.02
```

pierotofy, 2022-02-17:
Closing, assuming issue not due to WebODM.

closed 2022-02-17 by pierotofy