WebODM/WebODM #1104: V1.9.11 update WebODM dashboard goes blank when selecting task after update

justgeekit, 2021-11-30:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

(https://whatismybrowser.com/w/52JCHDB)

### What is the problem?

V1.9.11 update WebODM dashboard goes blank when selecting task after update

### What should be the expected behavior?

When selecting a task it should expand the task options.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

When I go to the WebODM dashboard view tasks for a project and select a task the dashboard goes blank.

[image]
[image]

pierotofy, 2021-11-30:
Hey @justgeekit thanks for the report! Are you running this with docker? (via `./webodm.sh start` ?)

Minhom, 2021-12-01:
Maybe u can turn off cesiumION in Administration   
[image]

MrCare, 2021-12-01:
awesome! it works!!! @Minhom

justgeekit, 2021-12-01:
Yes, I am using docker via ./webodm.sh start.

pierotofy, 2021-12-01:
The Cesium Ion plugin has not been updated for years; I'm not sure why it broke (or when), but Cesium GS Inc. hasn't really been supporting this plugin, so I'm going to remove it at the next release.

If this plugin is important, I would urge people to get Cesium's team involved.

Archived plugins will now live on https://github.com/OpenDroneMap/WebODM-Plugins-Archive

closed 2021-12-01 by pierotofy