WebODM/WebODM #1110: Fails to run on Nixos

paddygoat, 2021-12-09:
I amended to shebang in the webodm.sh file to: #!/usr/bin/env bash
The script executed with the following errors. Please advise !!!! Thank you:

[paddygoat@nixos:~]$ cd WebODM

[paddygoat@nixos:~/WebODM]$ ./webodm.sh start
Checking for docker...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
Traceback (most recent call last):
  File "/nix/store/cjhimawmks6yiiw4mdfhk60vhiym3536-python3.8-urllib3-1.26.4/lib/python3.8/site-packages/urllib3/connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "/nix/store/cjhimawmks6yiiw4mdfhk60vhiym3536-python3.8-urllib3-1.26.4/lib/python3.8/site-packages/urllib3/connectionpool.py", line 394, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1256, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1302, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/transport/unixconn.py", line 43, in connect
    sock.connect(self.unix_socket)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/nix/store/h4x196l4p1fd4nbd6vkz0fkgvz0lq210-python3.8-requests-2.25.1/lib/python3.8/site-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/nix/store/cjhimawmks6yiiw4mdfhk60vhiym3536-python3.8-urllib3-1.26.4/lib/python3.8/site-packages/urllib3/connectionpool.py", line 755, in urlopen
    retries = retries.increment(
  File "/nix/store/cjhimawmks6yiiw4mdfhk60vhiym3536-python3.8-urllib3-1.26.4/lib/python3.8/site-packages/urllib3/util/retry.py", line 532, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/nix/store/cjhimawmks6yiiw4mdfhk60vhiym3536-python3.8-urllib3-1.26.4/lib/python3.8/site-packages/urllib3/packages/six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "/nix/store/cjhimawmks6yiiw4mdfhk60vhiym3536-python3.8-urllib3-1.26.4/lib/python3.8/site-packages/urllib3/connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "/nix/store/cjhimawmks6yiiw4mdfhk60vhiym3536-python3.8-urllib3-1.26.4/lib/python3.8/site-packages/urllib3/connectionpool.py", line 394, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1256, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1302, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/nix/store/32q6ryrb860sksdi1al5djg3pgcpq92l-python3-3.8.11/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/transport/unixconn.py", line 43, in connect
    sock.connect(self.unix_socket)
urllib3.exceptions.ProtocolError: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/api/client.py", line 214, in _retrieve_server_version
    return self.version(api_version=False)["ApiVersion"]
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/api/daemon.py", line 181, in version
    return self._result(self._get(url), json=True)
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/api/client.py", line 237, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "/nix/store/h4x196l4p1fd4nbd6vkz0fkgvz0lq210-python3.8-requests-2.25.1/lib/python3.8/site-packages/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/nix/store/h4x196l4p1fd4nbd6vkz0fkgvz0lq210-python3.8-requests-2.25.1/lib/python3.8/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/nix/store/h4x196l4p1fd4nbd6vkz0fkgvz0lq210-python3.8-requests-2.25.1/lib/python3.8/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/nix/store/h4x196l4p1fd4nbd6vkz0fkgvz0lq210-python3.8-requests-2.25.1/lib/python3.8/site-packages/requests/adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/nix/store/7i7ggchqdxv862jf32gigwrax1vkcksc-docker-compose-1.28.6/bin/.docker-compose-wrapped", line 9, in <module>
    sys.exit(main())
  File "/nix/store/7i7ggchqdxv862jf32gigwrax1vkcksc-docker-compose-1.28.6/lib/python3.8/site-packages/compose/cli/main.py", line 81, in main
    command_func()
  File "/nix/store/7i7ggchqdxv862jf32gigwrax1vkcksc-docker-compose-1.28.6/lib/python3.8/site-packages/compose/cli/main.py", line 198, in perform_command
    project = project_from_options('.', options)
  File "/nix/store/7i7ggchqdxv862jf32gigwrax1vkcksc-docker-compose-1.28.6/lib/python3.8/site-packages/compose/cli/command.py", line 60, in project_from_options
    return get_project(
  File "/nix/store/7i7ggchqdxv862jf32gigwrax1vkcksc-docker-compose-1.28.6/lib/python3.8/site-packages/compose/cli/command.py", line 152, in get_project
    client = get_client(
  File "/nix/store/7i7ggchqdxv862jf32gigwrax1vkcksc-docker-compose-1.28.6/lib/python3.8/site-packages/compose/cli/docker_client.py", line 41, in get_client
    client = docker_client(
  File "/nix/store/7i7ggchqdxv862jf32gigwrax1vkcksc-docker-compose-1.28.6/lib/python3.8/site-packages/compose/cli/docker_client.py", line 170, in docker_client
    client = APIClient(use_ssh_client=not use_paramiko_ssh, **kwargs)
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/api/client.py", line 197, in __init__
    self._version = self._retrieve_server_version()
  File "/nix/store/1xdc10gdh8gzi9vy120a71hql5ky8ixk-python3.8-docker-4.4.4/lib/python3.8/site-packages/docker/api/client.py", line 221, in _retrieve_server_version
    raise DockerException(
docker.errors.DockerException: Error while fetching server API version: ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))

[paddygoat@nixos:~/WebODM]$

pierotofy, 2021-12-09:
Hi @paddygoat,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-12-10 06:35) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-12-09:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2021-12-09 by pierotofy

joshinils, 2022-12-11:
> Could we move this conversation over to the forum at https://community.opendronemap.org? pray The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! +1

I seem to have the same Issue, where on the Forum can I find an answer to this?