WebODM/WebODM #1111: VARI GeoTIFF Raw map export error after latest updates

SeanHill89, 2021-12-10:
### How did you install WebODM (docker, installer, etc.)?

Paid installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Google Chrome

### What is the problem?

Hi there

I am having issues exporting the VARI maps in geotiff raw format after the latest updates. Before hand I could export the map as a single band grey map with the values at -1 and 1 for vegetation analysis in QGIS. Now I am only able to export the RGB geotiff. This does not work for use in vegetation analysis as the bands are defaulted to 0 255 in all RGB band which does not bare relevance to the indice values between -1 and 1 representing veg cover/health/soil. Every time i try to export the raw image rather than the RGB, a note comes up saying “band index 2 out of range (not in (1,))” and I am unable to export the data and the export option defaults back to geotiff RGB. Please help!! I am processing on “High resolution” setting. I have searched the topic in the forum but there are no matches to this topic. This happens with the other RGB indices (EXG, TGl etc) as well, cannot export the raw geotiff versions.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

I run the image data in the "high Resolution" preset without changing any parameters

See screenshot bellow image below:

[image]

See drive data set link:

https://drive.google.com/file/d/1IEpQU1tGqgPHU3zUD4CNU3uQ0yV10Uvh/view?usp=sharing

pierotofy, 2021-12-15:
Hey @SeanHill89 :hand: thanks for the report; I think the Google Drive link is protected? (I cannot access it)

SeanHill89, 2021-12-17:
> Hey @SeanHill89 <https://github.com/SeanHill89> ✋ thanks for the report;

Hi OpenDroneMap/WebODM Piero

It shouldn't be, I will have a look and see if I can rectify it

Kind Regards
Sean

SeanHill89, 2021-12-17:
Seems to be working ok, just have to download them

pierotofy, 2021-12-17:
Seems OK now, thanks!

waruthp, 2022-03-01:
> Seems to be working ok, just have to download them

Hello Sean,

I have faced the same issue as you mentioned. Please advise me too. Thank you.

waruthp, 2022-03-01:
I also faced VARI GeoTIFF Raw map export error after latest updates

pierotofy, 2022-03-01:
Thanks for reporting it, it should be fixed for good with https://github.com/OpenDroneMap/WebODM/commit/79ce07eb287786dbb6bbb7eec0352a28c82dc0ba

waruthp, 2022-03-01:
> Thanks for reporting it, it should be fixed for good with [79ce07e](https://github.com/OpenDroneMap/WebODM/commit/79ce07eb287786dbb6bbb7eec0352a28c82dc0ba)

Thank you I will try.

SeanHill89, 2022-03-01:
Thank you very much! Excellent software