WebODM/WebODM #1113: WebODM

gunnakaa, 2021-12-15:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************
### How did you install WebODM (docker, installer, etc.)?

[First I installed Docker then I got an tip from Saijin-Naib that I also ned to install SSL. After I follow this recipe; https://docs.opendronemap.org/installation/#install-on-ubuntu-debian]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[I use Linux Ubuntu 20.04 and firefox webrovser]

### What is the problem?

[~/WebODM$ ./webodm.sh restart
Checking for docker...   OK
Checking for docker-compose...   OK
Restarting WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.nodemicmac.yml down --remove-rphans cannot update snap namespace: cannot create symlink in "/etc/docker": existing file in the way 
snap-update-ns failed with code 1]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

I was working firstly to get Docker up and go then the SSL.  The strange thing is I can not  find any folder for docker any place on my drive only in the snap folder is an docker folder that is empty. But if I write docker in the terminal I will get an menu for docker and webodm says it is OK.

pierotofy, 2021-12-15:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2021-12-15 by pierotofy

gunnakaa, 2021-12-16:
Thanks for fast answer :) It is total fine I will put my question there :+1: