WebODM/WebODM #1117: 3D

consultorfelipe, 2021-12-21:
I performed the installation on my Macbook Air 100%, but the 3D model is not working, I also installed it on my windows and I know I need to enable virtualization, how do I do it on my mac now?

pierotofy, 2021-12-21:
Hi @consultorfelipe,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-12-22 10:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

consultorfelipe, 2021-12-21:
Como você instalou o WebODM (docker, instalador, etc.)?
Docker

Qual é o seu navegador e sistema operacional? (Copie / cole a saída de https://www.whatismybrowser.com/ )
safari

Qual é o problema?
3D não roda, sei que no Windows precisa habilitar a virtualização, mas como falo isso no mac? 

Como podemos reproduzir isso? (Quais etapas desencadeiam o problema? Quais parâmetros você está usando para o processamento? Incluir capturas de tela. Se você estiver tendo problemas para processar um conjunto de dados, deve incluir uma cópia do conjunto de dados carregado no Dropbox, Google Drive ou https://dronedb.app )
3D nao finciona

closed 2021-12-21 by pierotofy