WebODM/WebODM #1119: Add a security policy

JamieSlome, 2022-01-08:
Hey there!

I belong to an open source security research community, and a member (@srikanthprathi) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

pierotofy, 2022-01-08:
Hi @JamieSlome,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-01-09 18:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2022-01-09:
Done, thanks! https://github.com/OpenDroneMap/WebODM/blob/master/SECURITY.md

closed 2022-01-09 by pierotofy

JamieSlome, 2022-01-11:
@pierotofy - you can view the report directly here:

https://huntr.dev/bounties/586ac0c7-14ab-4502-b5af-bb3f4d768207/

It is private and only accessible to maintainers with repository write permissions.