WebODM/WebODM #1126: Different layers of Camera is no visible

galensonet1, 2022-01-24:
Hello,
I am using WebODM for a while now. I currently have version 1.9.11 installed on my server.

I would like to use this tool to share the web links with my clients, but I find some inconsistencies that I would like to know if I can solve:

In these sites with several tasks, we also cannot see the different layers of cameras, only the first one that we made appears.
I Attach a screen capture

pierotofy, 2022-01-24:
Hi @galensonet1,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-01-25 16:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2024-05-28:
This should be fixed with the latest version. Thanks for the report!

Feel free to re-open if the issue persists? :pray:

closed 2024-05-28 by pierotofy