WebODM/WebODM #1127: Lightning node is hung while resizing images

konradkern, 2022-01-25:
This may be simple like restarting the lightning server.  I have submitted the same job twice.  The job is large-ish (345 images) and it has a GCP file.    The processor has hung both times at the resizing images step.  I had to halt the job once and may not have allowed the server to close properly.  Can you help?

pierotofy, 2022-01-25:
Hi @konradkern,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-01-26 11:20) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2022-01-25:
Hey @konradkern :hand: if it's stuck at "resizing" then it's probably something stuck on your local machine, not on the Lightning server. Do you have enough free disk space?

konradkern, 2022-01-25:
> Hey @konradkern <https://github.com/konradkern> ✋ if it's stuck at

I thank you.  Yes, I have gobs of Ram and disk space.  I've run my computer
cleaner and restarted after I aborted the jobs.  I'm sure it will resolve
in time.

Konrad
ph 714-658-4972

closed 2023-08-21 by pierotofy