WebODM/WebODM #1131: webodm.sh should respect the WO_DEFAULT_NODES env variable by default

gromain, 2022-01-31:
When starting WebODM with the command line (with `./webodm.sh start`) and after configuring the `.env` file, it appears that a node is started even if the variable `WO_DEFAULT_NODES` is set to 0.

Probably just adding `default_nodes=$WO_DEFAULT_NODES` after the `.env` sourcing [here](https://github.com/OpenDroneMap/WebODM/blob/6a16b1593c43d32d35edb802bed54e3aa1a0b4d7/webodm.sh#L39) would be enough.

gromain, 2022-01-31:
damn gitissuebot you're annoying for such a simple issue.

### How did you install WebODM (docker, installer, etc.)?

docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

not relevant

### What is the problem?

See first post

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

See first post

pierotofy, 2022-01-31:
Good point, would you be interested in opening a pull request @gromain ? :pray:

gromain, 2022-01-31:
Sure can! Will look at it tomorrow! :-)

PR #1132 (2022-02-01): webodm.sh: remove default_nodes and respect WO_DEFAULT_NODES from .env
This fixes #1131 and harmonizes the way WO_DEFAULT_NODES is managed.

Also, I have a warning list from shellcheck for this script that I can fix while I'm at it. This will make the script more robust. For example it looks like the password does not support having spaces in it for now (see #1133 ).

```
Warning	Quote this to prevent word splitting. [SC2046]	3:16
Info	Not following: ./.env was not specified as input (see shellcheck -x). [SC1091]	32:8
Warning	Declare and assign separately to avoid masking return values. [SC215 [... cut]
commit 90dd3db1 (2022-02-01): webodm.sh: remove default_nodes and respect WO_DEFAULT_NODES from .env

This fixes #1131 and harmonizes the way WO_DEFAULT_NODES is managed.
closed 2022-02-01 by pierotofy