WebODM/WebODM #1133: webodm.sh: impossible to set a password with a space from command line

gromain, 2022-02-01:
### How did you install WebODM (docker, installer, etc.)?

docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

not relevant

### What is the problem?

Trying to set a password with a space in it changes it to only the first word.

Command and output is
```
webodm:~/WebODM# ./webodm.sh resetadminpassword something with a space
The following user was changed: user
Password changed!
```

[image]

Password is set to `something` instead of `something with a space`.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
Start a container, login and create a user and change the password. You can't log in if the password has a space in it.

pierotofy, 2022-02-01:
You probably need to add quotes to your password (does it work then?)

gromain, 2022-02-01:
It does, but I feel it would be better to handle this directly in the script. A good practice is to use [shellcheck](https://github.com/koalaman/shellcheck/) to catch issues with bash scripts. It runs automatically with my IDE and warned me against a possible issue [here](https://github.com/OpenDroneMap/WebODM/blob/9c78aa8f35e60a3b0f23e3eec55558233dfee762/webodm.sh#L484).

I just did some more check, even with quoting the password, it's not enough. We need to both quote the password and fix the script. I can probably do that too, with the rest of the shellcheck fixes (and add something to the help message to make it clear to users that the password should be double quoted).

PR #1135 (2022-02-01): webodm.sh: Fix the password with space issue
Closes #1133.

Also brings a lot of shellcheck suggestions.
commit 08b916bd (2022-02-01): webodm.sh: shellcheck fixes and closes #1133
closed 2022-02-01 by pierotofy