WebODM/WebODM #1144: There are no images in /var/~...Make sure that your project path and dataset name is correct

Ryobidelacroix, 2022-02-14:
****************************************
### How did you install WebODM (docker, installer, etc.)?

Docker, 4.4.2

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 98 on macOS (Monterey)

### What is the problem?

"Cannot Process Dataset" 
issue occurs when processing submodels. 

[ERROR]   There are no images in /var/www/data/4615f86d-3f0a-4ad9-80ba-4f5ec5df17db/submodels/submodels/images! Make sure that your project path and dataset name is correct. The current is set to: /var/www/data/4615f86d-3f0a-4ad9-80ba-4f5ec5df17db/submodels/submodels

```
===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "/code/stages/odm_app.py", line 94, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 346, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 327, in run
self.process(self.args, outputs)
File "/code/stages/splitmerge.py", line 164, in process
system.run(" ".join(map(double_quote, map(str, argv))), env_vars=os.environ.copy())
File "/code/opendm/system.py", line 106, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

```
### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Options: auto-boundary: true, dsm: true, dtm: true, pc-tile: true, rerun-from: odm_report, split: 300, split-overlap: 20, verbose: true

Full console log (verbose):

```
[INFO]    DTM is turned on, automatically turning on point cloud classification
[INFO]    Initializing ODM 2.7.2 - Mon Feb 14 08:37:04  2022
[INFO]    ==============
[INFO]    auto_boundary: True
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    cog: True
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    debug: False
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    depthmap_resolution: 640
[INFO]    dsm: True
[INFO]    dtm: True
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: high
[INFO]    feature_type: sift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gps_accuracy: 10
[INFO]    ignore_gsd: False
[INFO]    matcher_neighbors: 8
[INFO]    matcher_type: flann
[... 220 lines cut ...]
[INFO]    ==============
[INFO]    Running dataset stage
[INFO]    Loading dataset from: /var/www/data/4615f86d-3f0a-4ad9-80ba-4f5ec5df17db/submodels/submodels/images
[ERROR]   There are no images in /var/www/data/4615f86d-3f0a-4ad9-80ba-4f5ec5df17db/submodels/submodels/images! Make sure that your project path and dataset name is correct. The current is set to: /var/www/data/4615f86d-3f0a-4ad9-80ba-4f5ec5df17db/submodels/submodels

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "/code/stages/odm_app.py", line 94, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 346, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 327, in run
self.process(self.args, outputs)
File "/code/stages/splitmerge.py", line 164, in process
system.run(" ".join(map(double_quote, map(str, argv))), env_vars=os.environ.copy())
File "/code/opendm/system.py", line 106, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/

```

pierotofy, 2022-10-19:
This should be fixed in the latest version of ODM/WebODM; please try to update and see if the issue persists? :pray: 

If it's still there, please share the dataset/images you're using along with steps on how to reproduce?

closed 2022-10-19 by pierotofy

Stefal, 2022-12-20:
Hi!
I have the same problem, with ODM 3.0.2 (Installed with docker on Debian)

Options : auto-boundary: true, camera-lens: equirectangular, dem-resolution: 2.0, dsm: true, gps-accuracy: 0.1, max-concurrency: 10, orthophoto-resolution: 2.0, pc-quality: high, pc-tile: true, rerun-from: odm_postprocess, sfm-algorithm: triangulation, split: 400

```
===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "/code/stages/odm_app.py", line 81, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 386, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 365, in run
self.process(self.args, outputs)
File "/code/stages/splitmerge.py", line 176, in process
system.run(" ".join(map(double_quote, map(str, argv))), env_vars=os.environ.copy())
File "/code/opendm/system.py", line 109, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====
```

The full log is too long, here is an extract :

```
[INFO]    Initializing ODM 3.0.2 - Mon Dec 19 17:15:35  2022
[INFO]    ==============
[INFO]    3d_tiles: False
[INFO]    align: None
[INFO]    auto_boundary: True
[INFO]    auto_boundary_distance: 0
[INFO]    bg_removal: False
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: equirectangular
[INFO]    cameras: {}
[INFO]    cog: True
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 2.0
[INFO]    dsm: True
[INFO]    dtm: False
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: high
[INFO]    feature_type: sift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gps_accuracy: 0.1
[INFO]    ignore_gsd: False
[INFO]    matcher_neighbors: 0
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 10
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 10000
[INFO]    name: a352e27a-6853-4293-b2a2-681e369f67d4
[INFO]    no_gpu: False
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_kmz: False
[INFO]    orthophoto_no_tiled: False
[... 230 lines cut ...]
[INFO]    ==============
[WARNING] Directory submodels does not exist. Creating it now.
[INFO]    Running dataset stage
[INFO]    Loading dataset from: /var/www/data/a352e27a-6853-4293-b2a2-681e369f67d4/submodels/submodels/images
[ERROR]   There are no images in /var/www/data/a352e27a-6853-4293-b2a2-681e369f67d4/submodels/submodels/images! Make sure that your project path and dataset name is correct. The current is set to: /var/www/data/a352e27a-6853-4293-b2a2-681e369f67d4/submodels/submodels

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "/code/stages/odm_app.py", line 81, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 386, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 365, in run
self.process(self.args, outputs)
File "/code/stages/splitmerge.py", line 176, in process
system.run(" ".join(map(double_quote, map(str, argv))), env_vars=os.environ.copy())
File "/code/opendm/system.py", line 109, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/
```