WebODM/WebODM #1145: Update Install - Failed - WebODM will not work now

RiverSonic, 2022-02-18:
### How did you install WebODM (docker, installer, etc.)?

Bought and downloaded apprx. year ago. Had a docker (hyper-V) at start then moved to all windows. I have no computer skills at all. 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Edge 98 on Windows 10

### What is the problem?

Haven't used app for awhile. Fired it up just now to update before starting some new projects. Figured there was probably an update available. Downloaded update and started install. Asked to put into existing WebODM folder, said yes. Failed on "Closing applications" dialoge box. Let it sit for awhile. Forced close. Restarted PC. Tried to Open WebODM received following error...

[image]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

If I try to start using icon...fails. Don't get very far. 

[image]

RiverSonic, 2022-02-19:
> Mm, strange. Have you tried re-installing the application?

Link to best guide for that?

RiverSonic, 2022-02-22:
>    1. Uninstall WebODM
>    https://support.microsoft.com/en-us/windows/uninstall-or-remove-apps-and-programs-in-windows-10-4b55f974-2cc6-2d2b-d092-5905080eaf98
>    2. Re-run the WebODM setup. That's it!

Afternoon,

I uninstalled and re-installed.

Now saying I need to have the containers running on Docker. I am pretty
sure it went to a windows application only a few months back and didn't
need the docker.

Can't get past this point now.

Russell

RiverSonic, 2022-02-22:
Error message when trying to initiate container.

docker: error during connect: This error may indicate that the docker
daemon is not running.: Post
"http://%2F%2F.%2Fpipe%2Fdocker_engine/v1.24/containers/create":
open //./pipe/docker_engine: The system cannot find the file specified.
See 'docker run --help'.

pierotofy, 2022-02-23:
Double-check you're running the latest/newer setup? The new setup does not require docker or containers. Reach out to hello@uav4geo.com if you need a link to the latest setup.

closed 2022-02-25 by pierotofy