WebODM/WebODM #1146: Possible Bug: img_path malformed in OpenAerialMap plugin

Saijin-Naib, 2022-02-25:
### How did you install WebODM (docker, installer, etc.)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Microsoft Edge Beta (flatpak), Alpine Linux 3.15 Edge

### What is the problem?
OpenAerialMap fails to work properly under recent versions of WebODM, citing "Internal Server Error":  
[image]

Salient console log entries:  
```
webapp      | WARNING Cannot get SETTINGS key from context. Something's wrong in settings_image_url.
webapp      | WARNING Cannot get SETTINGS key from context. Something's wrong in settings_image_url.
webapp      | WARNING Cannot get SETTINGS key from context. The footer will not be displayed.
webapp      | ERROR Internal Server Error: /api/plugins/openaerialmap/task/76ec544b-eba4-4614-a0dd-28dbb7e002a5/info
webapp      | Traceback (most recent call last):
webapp      |   File "/usr/local/lib/python3.9/dist-packages/django/core/handlers/exception.py", line 34, in inner
webapp      |     response = get_response(request)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/django/core/handlers/base.py", line 126, in _get_response
webapp      |     response = self.process_exception_by_middleware(e, request)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/django/core/handlers/base.py", line 124, in _get_response
webapp      |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
webapp      |   File "/webodm/app/plugins/views.py", line 59, in api_view_handler
webapp      |     return view(request, *args, **kwargs)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/django/views/decorators/csrf.py", line 54, in wrapped_view
webapp      |     return view_func(*args, **kwargs)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/django/views/generic/base.py", line 68, in view
webapp      |     return self.dispatch(request, *args, **kwargs)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/rest_framework/views.py", line 495, in dispatch
webapp      |     response = self.handle_exception(exc)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/rest_framework/views.py", line 455, in handle_exception
webapp      |     self.raise_uncaught_exception(exc)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/rest_framework/views.py", line 492, in dispatch
webapp      |     response = handler(request, *args, **kwargs)
webapp      |   File "/webodm/coreplugins/openaerialmap/api.py", line 62, in get
webapp      |     img_path = os.path.join(settings.MEDIA_ROOT, img.path())
webapp      | AttributeError: 'NoneType' object has no attribute 'path'
```

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
Enable OpenAerialMap plugin on latest WebODM docker versions:  
```
docker images
REPOSITORY                   TAG       IMAGE ID       CREATED        SIZE
opendronemap/nodeodm         latest    89b45eed4653   26 hours ago   1.77GB
opendronemap/webodm_webapp   latest    ed8b980879e1   7 days ago     1.6GB
redis                        latest    f1b6973564e9   4 weeks ago    113MB
opendronemap/webodm_db       latest    e40c0f274bba   5 months ago   695MB
```
Configure OpenAerialMap with API Key
Save API Key
Go to Dashboard
Expand Project, expand Task
Note that an infobox with "Internal Server Error" pops up automatically after a few seconds
Note that clicking "Share to OAM" does not trigger any actions

pierotofy, 2022-02-25:
I'm guessing this task was imported and not processed (by uploading images?) Looks like an edge case that should be fixed.

Saijin-Naib, 2022-02-25:
Yes, exactly, an all.zip that I sideloaded since, as you can imagine, processing on 4GB RAM with a Celeron N3450 isn't exactly friendly haha.

closed 2022-02-28 by pierotofy