WebODM/WebODM #1149: WebODM 1.9.11 for windows 10 64 bit cannot process dataset

waruthp, 2022-03-01:
### How did you install WebODM (docker, installer, etc.)?

paid installer v. 1.9.11

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

win 10 64 bit

### What is the problem?

this version cannot process the dataset of NIR (Near Infrared) images (but the docker version can do)

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

normal upload and process

https://drive.google.com/file/d/1m504UN1ZsFS-E38qztShodrEoJFFWc-E/view?usp=sharing

waruthp, 2022-03-13:
Need help please

pierotofy, 2022-07-08:
Hi @waruthp :hand: I've just tried to process this dataset and seemed to work fine with the latest version?

[image]

Perhaps try updating; if the problem persists, feel free to re-open including a copy of your task log? :pray:

closed 2022-07-08 by pierotofy