WebODM/WebODM #1150: Cannot process dataset

jeffersonpimenta, 2022-03-02:
Hi, i got WebODM running on a old dell server running without problems for almost a year.

Today i tried to update to the new version and got the following error:

[image]

When i update i rebuilt e whole software because my processor dont have some functions, following the article i wrote myself:

https://community.opendronemap.org/t/unsuported-cpu-intel-xeon-e5410/7560/9

I tried to update the python path, but got no sucess, following:

[image]

Any ideas? Thanks!

pierotofy, 2022-03-02:
Hi @jeffersonpimenta,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-03-03 08:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

jeffersonpimenta, 2022-03-02:
@pierotofy Sorry, i've installed via docker on a Dell 2950. My system is based on Debian.

When i first try to process i got the message that my processor is not suported. After i rebuilt like i wrote in this link:

https://community.opendronemap.org/t/unsuported-cpu-intel-xeon-e5410/7560/9

I got the message that opensfm was not found. After a quick search i got that the problem maybe was caused by a missing link on a python path, i search where opensfm was and put it on path without success.

jeffersonpimenta, 2022-03-02:
I got a solution, here the steps:

Enter container

docker exec -it "containerid" /bin/bash

navigate to build folder

cd /code/SuperBuild/build

Rebuild
make

After that i got one error message, it was a importation error, i fix according this link:

https://github.com/smicallef/spiderfoot/issues/1124

nano /usr/local/lib/python3.9/dist-packages/networkx/algorithms/dag.py

remove from fractions import gcd
and insert import math

After that everything works fine

[image]

closed 2022-03-02 by jeffersonpimenta