WebODM/WebODM #1151: Possible Bug: OpenAerialMap Plugin UI Button Missing, Responisve Reflow

Saijin-Naib, 2022-03-03:
### How did you install WebODM (docker, installer, etc.)?
Docker
```
docker images
REPOSITORY                   TAG       IMAGE ID       CREATED        SIZE
opendronemap/webodm_webapp   latest    b0abd6f064fa   47 hours ago   1.6GB
redis                        latest    8f8e3073a3f6   47 hours ago   113MB
opendronemap/nodeodm         latest    f6b1eba9f8e9   4 days ago     1.77GB
opendronemap/webodm_db       latest    e40c0f274bba   6 months ago   695MB
```

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Microsoft Edge Beta, Flatpak, Version 99.0.1150.30
Alpine Linux 3.15 Edge, x64

### What is the problem?
The OpenAerialMap plugin does not display the UI button for publishing after the WebODM upgrade to v1.9.12:  
[image]

In addition, the Responsive reflow of the UI buttons for the Tasks seems to be splitting the row into multiple rows early, when sufficient width is available for more buttons in a single row.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
Run version as above. Enable/configure all plugins as per below:  
[image]

This UI behavior is consistent with WebODM for Windows native v1.9.12 Build 55, as well.

Saijin-Naib, 2022-03-03:
[image]
I think if possible, we should be wrapping lines on whole words, not per-letter.

pierotofy, 2022-03-03:
For the button, I think those tasks have been imported, not processed, so they have no images available to share to OAM (that's why the button does not display).

I've fixed the word-break, I had changed it with the last update thinking it would help format longer error messages, but didn't catch the issue with other strings and the awkward breaks, so I just reverted it.

Saijin-Naib, 2022-03-03:
For that specific machine/instance, that is an imported task.
For my WebODM for Windows native instance, it shows the same behavior for tasks that were mixed processed locally and on Lightning, with a data directory spanning back... oof, long time.

Saijin-Naib, 2022-03-03:
WebODM Manager for Windows (Docker, WSL2):  
[image]

All of these tasks were processed locally on this machine/instance, but OAM plugin doesn't show for them. These were all Tasks started via the Cloud Import plugin, contacting the GitHub repositories directly.

pierotofy, 2022-07-22:
This should be fixed, if not, feel free to re-open! :pray:

closed 2022-07-22 by pierotofy