WebODM/WebODM #1152: Possible Bug: Task Completion Time setpoint from prior task time in Queue

Saijin-Naib, 2022-03-03:
### How did you install WebODM (docker, installer, etc.)?
Docker
```
docker images
REPOSITORY                   TAG       IMAGE ID       CREATED        SIZE
opendronemap/webodm_webapp   latest    b0abd6f064fa   47 hours ago   1.6GB
redis                        latest    8f8e3073a3f6   47 hours ago   113MB
opendronemap/nodeodm         latest    f6b1eba9f8e9   4 days ago     1.77GB
opendronemap/webodm_db       latest    e40c0f274bba   6 months ago   695MB
```

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Microsoft Edge Beta, Flatpak, Version 99.0.1150.30
Alpine Linux 3.15 Edge, x64

### What is the problem?
When processing multiple tasks on a single node in a queue, the subsequent tasks' processing times begin at something like 33s less than the final processing time of the preceeding task.

So:  
Task 1: Starts at 0:00
Task 1: Finishes at 5:00
Task 2: Starts at 4:27
Task 2: Finishes at 10:00
Task 3: Starts at 09:27
Task 3: Finishes at 15:00

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
My computer was laggy enough that I missed both transitions, but these are the same tasks in this screenshot for reference:  
[image]

Task 1 to Task 2 Transition:  
https://user-images.githubusercontent.com/19295950/156492244-2526391e-3904-44d1-a667-bd2ce6cddc06.mp4

Task 2 to Task 3 Transition:  
https://user-images.githubusercontent.com/19295950/156492283-3d0d390c-b17a-441b-805a-593698533be2.mp4

Saijin-Naib, 2022-03-03:
Test 1/3:
[image]
[Test-1-3-report.pdf](https://github.com/OpenDroneMap/WebODM/files/8174833/Test-1-3-report.pdf)

Test 2/3:  
[image]
[Test-2-3-report.pdf](https://github.com/OpenDroneMap/WebODM/files/8174834/Test-2-3-report.pdf)

Test 3/3:  
[image]
[Test-3-3-report.pdf](https://github.com/OpenDroneMap/WebODM/files/8174835/Test-3-3-report.pdf)

Saijin-Naib, 2022-03-03:
WebODM for Windows native v1.9.12 Build 55:  
Initial State:  
[image]

Final State:  
[image]

Same behavior as the Dockerized versions in regards to enqueued jobs starting with a non-zero task processing time.

Saijin-Naib, 2022-06-10:
Still valid for WebODM for Windows native v1.9.14 Build 61 and current Docker:  
[image]

```
docker images
REPOSITORY                     TAG       IMAGE ID       CREATED        SIZE
redis                          latest    dd8125f93b94   2 days ago     117MB
opendronemap/webodm_webapp     latest    e0ce4eafc83e   2 days ago     1.63GB
opendronemap/nodeodm           latest    65ed367c5bf2   3 days ago     1.8GB
opendronemap/webodm_db         latest    e40c0f274bba   9 months ago   695MB
docker/desktop-reclaim-space   latest    acaa0a8bbb14   2 years ago    101kB
```

pierotofy, 2025-03-11:
This seems to be due to NodeODM's logic, which uses creationTime (rather than startTime) to compute the total number of seconds elapsed. So if a task is in queue for 6 minutes, the timer will jump to 6 minutes at the start.

pierotofy, 2025-03-11:
Ref: https://github.com/OpenDroneMap/NodeODM/pull/233

pierotofy, 2025-03-12:
Same issue: #904

closed 2025-07-30 by smathermather