WebODM/WebODM #1153: Possible Bug: No Responsive Reflow on GCP Editor Pro "Resume Work on GCP File"

Saijin-Naib, 2022-03-09:
### How did you install WebODM (docker, installer, etc.)?
Stand-Alone (Windows)
WebODM Plugin (Docker, Windows native)

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Verified in Edge, Firefox
Windows 11, Alpine Linux 3.15 Edge

### What is the problem?
The "Resume Work on GCP File" page of GCP Editor Pro does not Responsive Reflow like the other pages do.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
Size the window fairly small, and observe how the Examples/Format Specification groups reflow on Create GCP File From CSV:  
[image]

Observe that those groups disappear on "Resume Work on GCP File":  
[image]

Video of the behavior:  
https://user-images.githubusercontent.com/19295950/157536532-cffea699-b8da-4853-95d7-b5c4fa76e84d.mp4

Saijin-Naib, 2022-03-09:
WebODM Docker (1369x768 host resolution:  
[image]

[image]

pierotofy, 2025-03-30:
No bandwidth for this at the moment.

closed 2025-03-30 by pierotofy

Saijin-Naib, 2025-03-30:
Saaad, but understood.