WebODM/WebODM #1155: Possible bug: Basic exif-tags not read after resizing images.

atobiese, 2022-03-15:
Hi! 
For some datasets, the exit-tags are not readable after image resizing.

[WARNING] Cannot read basic EXIF tags for /var/www/data/116a5dfd-a189-4e01-8cdb-5c53758ebbee/images/DSC00430.JPG: string index out of range

The try catch appears in the photo.py script, openmv. I have not been able to check why the tags are modified.  
It is correctly read when no resizing is done. I have seen this using a workflow involving explicitly tagging images from RTK-gps data. 
I will be able to share images, but will have to find a few that reproduces the errors, and may take a bit of time.

pierotofy, 2022-03-15:
Hi @atobiese,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-03-16 04:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

atobiese, 2022-03-24:
The following external bug fixes the error. 
https://github.com/atobiese/Piexif/commit/a58dee150c574c1c42b428631249b51e2151331f

It's located in [Piexif](https://github.com/hMatoba/Piexif) and the error appears when parsing exif data during loading (which is then dumped back after resizing in ODM).  
photo.py requests GPSLatitudeRef and GPSLongitudeRef for decimal degree calculation.
https://github.com/OpenDroneMap/ODM/blob/2e87505ce7e026c8fcee527a6524cdcbb7a6ec09/opendm/photo.py#L210

The problem is that piexif seems stale. So the question is how to resolve this. 
 
The problem occurs as these values were erased, none of the image sets failed (georeferencing) after resizing.

pierotofy, 2022-03-25:
Ah, good find! Thanks for looking into this :pray: 

I would recommend to open a pull request in the piexif repository; then once a new version for piexif is released we can upgrade it in WebODM. That would be the easiest way.

atobiese, 2022-03-25:
Sounds like a good plan:) I've opened the PR and hopefully it will get merged in soon.

closed 2025-11-11 by pierotofy