WebODM/WebODM #1158: Multispectral images strange stitching DJI P4M

simtrax, 2022-03-21:
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Brave, Pop! OS

### What is the problem?
[image]

I have tried processing two different datasets with WebODM taken by the DJI Phantom 4 Multispectral drone.
I have uploaded only the TIF images and not the RGB images that are stored in JPG files.

As you can see in the image the values in certain parts of the image are very strange.
My guess is maybe ODM mixes images from different bands?

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

The only parameters I selected for processing was "Multispectral" in the dropdown.
Unfortunately the dataset is 11 Gb (~2800 images). But if it is needed I will upload it.

simtrax, 2022-03-24:
@pierotofy Can you give any pointers on where to start looking if one were to check this issue out?
I'm a dev by profession but haven't checked out the ODM code base. So any pointers would be interesting.

simtrax, 2022-04-03:
@pierotofy I have now tried with more datasets and end up with similar results every time.
I have uploaded the above mentioned dataset, use the link below to download.
Will keep it there for a few days.

[Dataset, zip ~8 Gb](https://www.dropbox.com/s/y8j6bgqn1sju2uy/phantom%204%20multispectral.zip?dl=0)

If there's near solution in sight, please let me know. In that case I will have to look for other software to use.
And as I said, if there's anything I can do to help, let me know.

pierotofy, 2022-04-03:
Hey @simtrax :hand: thanks for sharing the dataset, I'm unsure of what the cause of this issue might be; my suggestion would be to isolate a small test case with as few images as possible (that will save you time while debugging), then check the various models generated in `odm_texturing_25d` (there's going to be one for each band) using meshlab (you might need to manually convert the texture files from tif to jpeg to open them). Check the point cloud. Check for band alignment issues. Check the log for warnings.

illxi, 2022-04-09:
i am having the same problem.

pierotofy, 2022-05-04:
Hey @goodshort :hand:

> odm_texturing_25d => Do you mean the folder odm_texturing when downloading all assets?

Nop, there's a separate model (2.5D). I'd recommend to run the pipeline directly from ODM to see the differences (this model is not included in the "all assets" archive from WebODM/NodeODM).

> What is your recommended way to convert tif files to jpeg?

GDAL.

> Any tuto/video/series that would explain how to review the band alignment issues?

Sorry I can't think of any good resource.

pierotofy, 2022-06-25:
Related: https://community.opendronemap.org/t/dji-p4m-segmented-orthophoto/12039/15

The patches from the screenshots are due to the `texturing-skip-global-seam-leveling` option, see the forum thread for a more in-depth discussion of the issue. Also https://github.com/OpenDroneMap/ODM/issues/1483

pierotofy, 2022-10-19:
We've made some improvements as of late with multispectral imagery and radiometric calibration. Is this still an issue?

simtrax, 2022-11-15:
@pierotofy Thanks for the answer.
It is late in the season now so there's not the need to collect images right now.
During spring I will venture out again and try it out. 

If you want to close the issue that's fine by me. If the issue is still there during spring I'll probably make another issue.

closed 2022-11-15 by pierotofy