WebODM/WebODM #1160: WebODM 9.1.12 bld 55 on win11 :"sending images to processing node " is hanging.

henrikhagenolesen, 2022-03-26:
### How did you install WebODM (docker, installer, etc.)?
WebODM standard installer (License fee) installed on March 20th.
Have been trying even since to get data processing to start.  WeBODM never gets beyond "sending images.."

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Firefox 98.0.2 (64-bit)  (or Edge ) 
Windows 11 ver. 21H2 bld 22000.556
Hardware :  MS surface 7 pro: 16 Gb RAM;  >100gb free SSD space ;  Intel(R) Core(TM) i7-1065G7 CPU @ 1.30GHz   1.50 GHz

### What is the problem?
Web ODM never starts to process 37 images from DJI drone  
Local node (IP 127.0.0.1 port 29301 ) never reach processing stage - on message "sending images to processing..
Attempt to use to Lightning node shows same "result" : hangs on "sending images" ...

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? 
No processing steps are ever reached.

pierotofy, 2022-03-26:
Hey @henrikhagenolesen :hand: could you attach the `app.log` file that you get after trying to process a job? (You can access it via Tools -- View Logs). Don't restart the app after processing the job, just open WebODM, try to process the job, make a copy of the log file (without closing WebODM) and attach it to this thread? (You might need to .zip it).

henrikhagenolesen, 2022-03-28:
Hello Piero,

Thank you for quick response.

I attach the latest app.log for your perusal

 

Looking forward to hearing from you.

 

/Henrik H.O

pierotofy, 2022-03-28:
You might need to visit https://github.com/OpenDroneMap/WebODM/issues/1160 and attach the log file to the discussion by dragging and dropping it (I didn't receive the attachment).

henrikhagenolesen, 2022-03-29:
[app.log](https://github.com/OpenDroneMap/WebODM/files/8370846/app.log)

henrikhagenolesen, 2022-03-29:
Hello Piero
Thank you
The app.log file should not be avaliable in this thread- see previous post.
/Henrik H.O

henrikhagenolesen, 2022-03-30:
Hello Piero
Thank you
The app.log file is avaliable in this thread- see previous post.
[app.log](https://github.com/OpenDroneMap/WebODM/files/8370846/app.log)
/Henrik H.O

pierotofy, 2022-04-03:
Try to check the contents of your `c:\windows\system32\drivers\etc\hosts`? For some reason the program is failing to initialize properly and the log seems to hint that there might be a wrong entry in that file.

You can use these instructions to reset the file to defaults: https://support.microsoft.com/en-us/topic/how-to-reset-the-hosts-file-back-to-the-default-c2a43f9d-e176-c6f3-e4ef-3500277a6dae

henrikhagenolesen, 2022-04-04:
Hello Piero
Thank you for the suggestion. 
The host file is in its default form, with no entries in it. Only the original MS intro header.  I attach the existing host file (for this purpose renamed to hosts.txt
[hosts.txt](https://github.com/OpenDroneMap/WebODM/files/8408869/hosts.txt).  

Meanwhile, FYI:  I have also installed WebODM on an old Linux (Ubuntu) PC (also a MS surface, but older and only 4GB ram) and this installation (Docker-based) is processing the same test dataset without problems -- and may I add -- with very nice results. But I still need to get the Windows installation up and running...

jlmaurer, 2023-02-17:
@pierotofy @henrikhagenolesen were you ever able to resolve this issue? I'm having the same issue on the same OS, on a Precision workstation, with the latest version of the paid Windows installer and local processing. 41 images never gets past sending to processor, also tried a couple different projects with same results. Log file attached. Tried both default and fast ortho processing modes and same results.

[app.log](https://github.com/OpenDroneMap/WebODM/files/10768240/app.log)

pierotofy, 2024-09-28:
As this issue doesn't seem to be a problem with WebODM per-se, but rather a problem with either the environment (security software, AV, firewall, enterprise settings), there's not much that can be done on the WebODM side. If anyone encounters this issue in 2024 with the WebODM Windows Installer, I'd advise to contact OpenDroneMap's support for further troubleshooting.

closed 2024-09-28 by pierotofy