WebODM/WebODM #1161: WebODM 1.9.13 : localhost node always "offline"

Majstor, 2022-03-26:
In my install (Fedora 35, using podman), WebODM constantly sees nodeodm as offline, and can't do any work. But if i check output, i can see the node is running :

```
$ ./webodm.sh start
Checking for podman...   OK
Checking for podman-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

podman-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
['podman', '--version', '']
using podman version: 3.4.4
** merged:
{
 "_dirname": "/home/majstor/WebODM_latest",
 "version": "2.1",
 "volumes": {
   "dbdata": null,
   "appmedia": null
 },
 "services": {
   "db": {
     "image": "opendronemap/webodm_db",
     "container_name": "db",
     "ports": [
       "5432"
     ],
     "volumes": [
       "dbdata:/var/lib/postgresql/data:Z"
     ],
     "restart": "unless-stopped",
     "oom_score_adj": -100
   },
   "webapp": {
     "image": "opendronemap/webodm_webapp",
     "container_name": "webapp",
     "entrypoint": "/bin/bash",
     "command": "-c \"/webodm/wait-for-postgres.sh db /webodm/wait-for-it.sh -t 0 broker:6379 -- /webodm/start.sh\"",
     "volumes": [
       "appmedia:/webodm/app/media:z"
     ],
     "ports": [
       "8000:8000"
     ],
     "depends_on": [
       "db",
       "broker",
       "worker",
       "node-odm"
     ],
     "environment": {
       "WO_PORT": null,
       "WO_HOST": null,
       "WO_DEBUG": null,
       "WO_BROKER": null,
       "WO_DEV": null,
       "WO_DEFAULT_NODES": null
[... 228 lines cut ...]
INFO Registered [coreplugins.diagnostic.plugin]
[2022-03-26 19:12:05,575: ERROR/MainProcess] beat: Connection error: Error 111 connecting to localhost:6379. Connection refused.. Trying again in 2.0 seconds...
INFO Registered [coreplugins.elevationmap.plugin]
INFO Registered [coreplugins.fullscreen.plugin]
INFO Registered [coreplugins.measure.plugin]
INFO Registered [coreplugins.openaerialmap.plugin]
INFO Registered [coreplugins.posm-gcpi.plugin]
ERROR Cannot connect to celery broker at redis://localhost. Make sure that your redis-server is running at that address: Error 111 connecting to localhost:6379. Connection refused.

Congratulations! └@(･◡･)@┐
==========================

If there are no errors, WebODM should be up and running!

Open a web browser and navigate to http://localhost:8000

NOTE: Windows users using docker should replace localhost with the IP of their docker machine's IP. To find what that is, run: docker-machine ip
[2022-03-26 19:12:07,578: ERROR/MainProcess] beat: Connection error: Error 111 connecting to localhost:6379. Connection refused.. Trying again in 4.0 seconds...
[2022-03-26 19:12:08,900: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379//: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 6.00 seconds...

[2022-03-26 19:12:11,583: ERROR/MainProcess] beat: Connection error: Error 111 connecting to localhost:6379. Connection refused.. Trying again in 6.0 seconds...
[2022-03-26 19:12:14,908: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379//: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 8.00 seconds...
```

pierotofy, 2022-03-26:
Hi @Majstor,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-03-27 15:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Majstor, 2022-03-27:
See nodeodm container's log :
```
info: Checking for orphaned directories to be removed...
info: Server has started on port 3000
info: Closing server
info: Exiting...
info: Authentication using NoTokenRequired
info: Listening on 0.0.0.0:6367 UDP for progress updates
info: Initialized 0 tasks
info: Checking for orphaned directories to be removed...
info: Server has started on port 3000
info: Closing server
info: Exiting...
info: Authentication using NoTokenRequired
info: Listening on 0.0.0.0:6367 UDP for progress updates
info: Initialized 0 tasks
info: Checking for orphaned directories to be removed...
info: Server has started on port 3000
info: Closing server
info: Exiting...
info: Authentication using NoTokenRequired
info: Listening on 0.0.0.0:6367 UDP for progress updates
info: Initialized 0 tasks
info: Checking for orphaned directories to be removed...
info: Server has started on port 3000
```

pierotofy, 2022-03-27:
Mm, we don't officially support podman, so you might need to make changes to WebODM/NodeODM to make it work (open a pull request?) :pray: 

Since this doesn't look like a bug, I'm going to close this for now.

closed 2022-03-27 by pierotofy

Majstor, 2022-03-27:
It has worked nice before this update. Any reasons ?

pierotofy, 2022-03-27:
I'm not really sure! We haven't changed our docker stuff in a while.

Majstor, 2022-04-02:
And ? No solution for me ?

pierotofy, 2022-04-03:
I'm sorry you feel that way.

https://docs.opendronemap.org/requesting-features/#how-to-request-features

We don't officially support podman.

You can either help us improve the software by opening a pull request or you could contribute in some other way to add support for podman.

Groupeffect, 2023-09-17:
Hi @Majstor,
I had the same problem with podman-compose and I just added the `env_file` attribute to the docker-compose.yml services `webapp` and `worker` like this:

```
# docker-compose.yml
...
webapp:
   ...
    env_file:
      - .env
   ...

worker:
   ...
    env_file:
      - .env
   ...
```
this way you set the `WO_BROKER` value to `redis://broker` from the `.env` file.
I hope this works for you, too.
Additionally you can add the prefix docker.io like: `image: docker.io/<image_name>`  to the docker-compose image paths.

paultrino, 2024-06-05:
@Groupeffect  You are my hero!!!! I also was having issues with Podman. Those two small changes in my docker-compose did it! 💪

paultrino, 2024-06-05:
@pierotofy Would it be ok to put @Groupeffect s suggestion into docker compose? I know podman is not supported, but it would just work out of the box for podman users with a minimal change. (PS, great book!)

paultrino, 2024-06-05:
@pierotofy  Piero, also a solid thank you to you as well for all your hard work on ODM 🍻. Your classiness, dedication, leadership, and contributions to GIS / FOSS / GiScience are greatly appreciated. Especially for drone newbs like myself. (Amazing book btw! highly recommended!!!!! [OpenDroneMap: The Missing Guide - 2nd Edition](https://odmbook.com/2)