WebODM/WebODM #1166: Shared datasets should not be removable by non-owners

pierotofy, 2022-04-12:
As a user I want to be able to unshare/remove projects that have been shared with me without them being deleted from the project owner’s account.

TimeScience, 2023-03-22:
@pierotofy can we re-open this? There are still some outstanding issues...

(1) If the user deletes the Shared project, it does disappear from their dashboard, but the popup text says "_All tasks, images and models associated with this project will be permanently deleted..._", not that they are unsharing... perhaps for Projects that are shared with people, the button should not be "delete" but "remove share" or some similar text and the popup should indicate that the user is removing that shared project from their dashboard but this won't impact the exitance of the project in the Project Owner's account?

(2) If the shared user duplicates the Project, the duplicated project appears in the Owner's dashboard, but not the shared user's dashboard. Options to fix this are (a) if a shared user duplicates the Project, all tasks and data are copied to their account and they have ownership of the new Project, or (b) either the user can't duplicate it if they don't own the data or they have to ask permission... (_both of these seem harder to implement so I'd go with (a)_). Ideally the owner of the data would also be notified that the user duplicated (ie copied) the data from that Project, but I assume this is also hard to implement so we can skip it for now.

(3) I was just able to delete a Task from a project that was shared read/write with me and the Task was deleted from the Project owner's account as well (ie actually deleted). If we assume sharing happens at the project level, then at the Task level we need to just disable people's ability to delete a task they don't own. I'd think the easiest implement this would be grey out the "Delete" button on Tasks that aren't owned by the user and then add a tooltip over the delete button telling them it is a shared task so they can't delete it. Alternatively, we could remove the delete button on shared tasks, but this will confuse people because there won't be an easy way to tell them why there is no Delete button, so greying it out seems like a better approach.

pierotofy, 2023-03-22:
1) Agree the message is a bit alarming. I'll change it.

2) Agree, the duplicate should be owned by the user that duplicated it.

3) Well, yes; you have "write" permissions, so you can delete tasks within a project. If you don't want a user to delete tasks within a project, you should assign "read" permissions to them. Tasks have no "owners" of their own (they just belong to a project).

TimeScience, 2023-03-22:
Thanks for the quick follow up.

re 3... yes I suppose that makes sense!

pierotofy, 2023-03-22:
Should be fixed with #1308. Please update and let me know if the issues persist :pray:

closed 2023-03-22 by pierotofy