WebODM/WebODM #1173: problems after update - no 3D models.

holoboss, 2022-04-23:
****************************************
### How did you install WebODM (docker, installer, etc.)?

[installer - update]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[windows 10 / 64   MS Edge]

### What is the problem?

[Hello, i used older version with Docker also from installer - without issues. But I recently updated to newest version and since then I can't see 3D model. nothing changed in my computer, same configuration. 2D maps are ok as before.  but 3D model does not show. Just a blank background. all tools on side are present, but i can't make 3D model appear. I checked project rendering options, made sure 3d model and other 3d options are ON (textured model, mesh, and so on). I tried several different configurations, various quality of Point Cloud rendering. i tried returning to default options too. Nothing. 3D model does not show up in any form. And when I upload same set of images to the DroneDB HUB online, i can create online 3D model without problem. Not sure what happens, and idea?
Many thanks!
Derek]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

[Type answer here]
[image]

pierotofy, 2022-05-05:
Hi @holoboss :hand: could you send us a copy of the `all.zip` (All Assets) archive to help us reproduce this? :pray:

holoboss, 2022-05-05:
Thank you. Here is download link to all.zip   (valid 1 week). i picked a very small project to keep size to minimum. 
https://we.tl/t-sltj4OzbUH

checked again before upload - can't see 3D on my WebODM.
Appreciate your help! Cheers
Derek

pierotofy, 2022-05-06:
Thanks Derek, this should have been fixed with and will be included in the next build/update.

In the meanwhile, if the model doesn't appear, try pressing the "Full Extent" button from the navigation menu:

[image]

closed 2022-05-06 by pierotofy

holoboss, 2022-05-06:
the Full Extent worked! Thank you! Sorry to bother you, appreciate your help!
Derek