WebODM/WebODM #1176: Possible Bug: DroneDB Integration Plugin needs update/fixing

Saijin-Naib, 2022-04-29:
### How did you install WebODM (docker, installer, etc.)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Alpine Linux Edge (3.15)

### What is the problem?
DroneDB Plugin can not populate any datasets:  
[image]
[image]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
Properly sign-in to DroneDB plugin and attempt to list any datasets you have.

Behavior confirmed in Docker and Windows install, issue report from Community:  
https://community.opendronemap.org/t/dronedb-integration-no-datasets-found/11353?u=saijin_naib

pierotofy, 2022-04-29:
Pinging @HeDo88TH

HeDo88TH, 2022-04-30:
> Pinging @HeDo88TH

Fixed 🚀

PR #1178 (2022-04-30): Fix DroneDB plugin
Addresses https://community.opendronemap.org/t/dronedb-integration-no-datasets-found/11353?u=saijin_naib 

Closes #1176
closed 2022-04-30 by pierotofy