WebODM/WebODM #1177: Possible Bug: Measurement Tools broken in Docker Latest

Saijin-Naib, 2022-04-29:
### How did you install WebODM (docker, installer, etc.)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Alpine Linux Edge (3.15)

### What is the problem?
When attempting to use the Measurement Tools on the View Map (2D), the viewport will shift drastically about -45deg (South-East) with each click.

Video:  
https://user-images.githubusercontent.com/19295950/166063071-9ac8431a-3558-419b-a8d1-014da0bd26aa.mp4

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
```
docker images
REPOSITORY                   TAG       IMAGE ID       CREATED        SIZE
redis                        latest    a10f849e1540   46 hours ago   117MB
opendronemap/webodm_webapp   latest    3d4dcaa0b5ea   2 days ago     1.62GB
opendronemap/nodeodm         latest    b4c50165f838   3 days ago     1.77GB
opendronemap/webodm_webapp   <none>    3c7ea741ec11   7 days ago     1.62GB
redis                        <none>    3c3da61c4be0   9 days ago     113MB
opendronemap/webodm_db       latest    e40c0f274bba   7 months ago   695MB
```

pierotofy, 2022-04-29:
Mm, strange, we haven't changed the code there in a while.

dmbworld, 2022-04-29:
@pierotofy I reported exactly the same problem in #1172, something has certainly changed and this tool is of significant importance to many.

Saijin-Naib, 2022-04-29:
@dmbworld Thanks!

So now we know it is present across both Chromium and Gecko rendering engines, so there goes my next line of inquiry...

cdcromwell, 2022-05-01:
I'm having the same issue on M1 Mac using Safari 15.4

pierotofy, 2022-05-02:
Well, this is *really* odd.

Our leaflet version was set to: `^1.3.1`, which means it should only receive updates for versions compatible with 1.3.1 (1.3.1 was released years ago, so I expect no updates should have ever been provided).

But during the latest build the map was upgraded silently to 1.8.0 (which is not compatible with some of our code).

I pinned the version to `1.3.1`, which should fix this.

pierotofy, 2022-05-02:
https://github.com/OpenDroneMap/WebODM/commit/5842de9d6ec58aebd9bbc6591fe2c8e226cfe778

Saijin-Naib, 2022-05-02:
Leaflet v1.8.x looks to offer many good performance and feature improvements that might make updating our code worthwhile for:  
https://github.com/Leaflet/Leaflet/releases/tag/v1.8.0

dmbworld, 2022-05-02:
Thanks @pierotofy and @Saijin-Naib, docker image has updated and I've updated my installation of WebODM, measurements are now working as before without any relocation of points.

closed 2022-05-02 by pierotofy

cdcromwell, 2022-05-02:
Works great now thanks