WebODM/WebODM #1179: WebODM not running after Windows 11 Pro and WebODM updates (04/05/2022)

SeanHill89, 2022-05-04:
WebODM not running after Windows 11 Pro and WebODM updates (04/05/2022)
Hi there WebODM geniuses

I did a Windows update a few weeks ago to windows 11 pro and did a WebODM update and now my maps do not run. I have a very beefy computer:
Processor: AMD Ryzen 9 5900HX with Radeon Graphics           3.30 GHz
RAM: 32.0 GB (31.4 GB usable)

Maps used to render very quickly. After updating, the processing node went offline so I uninstalled WebODM and reinstalled it a few times. This worked (sort of). Maps of 467 images that would run in half an hour now take 10 hours. I have the number of processors used by the computer set to max yet the CPU is only using 15% when running WebODM instead of 100% as it used to. How do I go about fixing this problem and how do I incorporate my GPU to work with the CPU when rendering maps to speed up the process (is it possible to use the GPU as well?). I have looked through the forum and found that others have had a similar issue and have tried the recommendations but nothing works. At the same time, the first time I load the images into WebODM, the process fails (see console for High resolution RGB and Multispectral consoles attached). I then reload the images and it will start processing but it take an extremely long time which it never did before. I ran the images through DJI TERRA (a far inferior software than WebODM in my opinion) and the maps run 7900 multispec images in 15 minutes and 227 images in 8 minutes.
I used the paid WebODM installer and my browser is Brave
The image overlap is 80% side 75% front.

Thank you for a fantastic program.
Kind regards

pierotofy, 2022-05-04:
Hi @SeanHill89,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-05-05 06:10) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2022-05-04:
Conversation already started here: https://community.opendronemap.org/t/webodm-not-running-after-windows-11-pro-and-webodm-updates-04-05-2022/11539/2

closed 2022-05-04 by pierotofy