WebODM/WebODM #1186: Order of adding images to task affects processing success or failure

Johnny555555, 2022-05-24:
Raised issue in relation to the discussion thread with the same name in the community forum:
https://community.opendronemap.org/t/order-of-adding-images-to-task-affects-processing-success-or-failure/11711

System info:
Ubuntu 20.04.4 LTS
512GB RAM
Dual Xeon 20-core CPUs
NVidia Telsa K80
Magnetic HDDs (3 RAID volumes: 2TB volume for OS, 2TB volume for Docker, 2TB volume for swap)
1.5TB swap file
Docker version 20.10.14, build a224086 (not using Docker Desktop)
WebODM 1.9.12
nodeodm:gpu

WebODM settings:
auto-boundary: true, debug: true, fast-orthophoto: true, gps-accuracy: 1, orthophoto-resolution: 1, rerun-from: dataset, resize-to: -1, verbose: true

4830 images at 20Mpx taken with a DJI P4P RTK
North-south nadir (2 directories containing images)
East-west nadir (2 directories containing images)
Oblique (45 degree gimbal pitch)  (2 directories containing images)

Directory Name     Qty Images
EW1                      999 images
EW2                      635 images
NS1                      999 images
NS2                      632 images
OB1                     999 images
OB2                     566 images

23 GCPs with 145 image cross-reference entries in GCP file
Datum: WGS84
Projection: UTM zone 56 south

Processing completes successfully if images are selected from the 6 directories in this order:
(NS1,NS2,EW1,EW2,OB1,OB2)
Processing fails after about 50 minutes if the images are selected in this order:
(EW1,EW2,NS1,NS2,OB1,OB2)

Johnny555555, 2022-05-25:
Just an FYI - I'm running more tests to clearly define the issue replication steps.

Johnny555555, 2022-05-25:
Selecting images in this order failed after 40min:
GCP file,EW1,EW2,NS1,NS2,OB1,OB2
Log file attache
[Test Server4-16 Shoalhaven EW1 EW2 NS1 NS2 OB1 OB2.txt](https://github.com/OpenDroneMap/WebODM/files/8770490/Test.Server4-16.Shoalhaven.EW1.EW2.NS1.NS2.OB1.OB2.txt)
d

Johnny555555, 2022-05-26:
Selecting images in this order completed successfully after 24h7min:
GCP file, NS1,EW1,OB1,NS2,EW2,OB2
Log file too large to attach so here's a OneDrive link to it (link expires June 30)
https://1drv.ms/t/s!AnbQHcX3KX2dhacUisUaTjJT2cSKKg?e=OPc3Bb

Johnny555555, 2022-05-28:
Selecting images in this order failed after 24h46min 
GCP file,NS1,NS2,EW1,EW2,OB1,OB2
with this error message:
[ERROR]   The program exited with a strange error code. Please report it at https://community.opendronemap.org
100 - done.

Log file too large to attach so here's a OneDrive link to it (link expires June 30)
https://1drv.ms/t/s!AnbQHcX3KX2dhacWSVheLWYpS0vVJw?e=6L9PPT

Johnny555555, 2022-05-29:
Repeated previous image selection order but with 3MPx images (instead of 20MPx images) selected in this order:
GCP file,NS1,NS2,EW1,EW2,OB1,OB2
Processing completed successfully after 19h48m
Log file too large again, so here's a OneDrive link to it (link expires June 30)
https://1drv.ms/t/s!AnbQHcX3KX2dhacx75l7gOdrYrYCfg?e=oxD43q

Mjankor, 2022-06-04:
I'll add I've also seen something similar. 

I recently started a drone mapping flight, but had to pause halfway through as someone wandered into the area. The second half of the job then had images with a different name series to the first. Flight was a single grid photogrammetry run with a Phantom RTK (connected to CORS). 

First set naming style
100_0005_0001.JPG
100_0005_0002.JPG
100_0005_0003.JPG
100_0005_0004.JPG
100_0005_0005.JPG
100_0005_0006.JPG
100_0005_0007.JPG
100_0005_0008.JPG

Second set (after the pause and restart)
100_0006_0001.JPG
100_0006_0002.JPG
100_0006_0003.JPG
100_0006_0004.JPG
100_0006_0005.JPG
100_0006_0006.JPG
100_0006_0007.JPG

Processing all of these files together resulted in a failure. Renaming them into one continuous series of filenames (Image001, Image 002, etc) and the job processed smoothly.

Johnny555555, 2022-06-05:
Interesting.
To avoid duplicate filenames I had prefixed my image file names according to the flight orientation and nadir vs oblique angle, i.e.
NS1_DJI_0001.JPG to NS1_DJI_0999.JPG
NS2_DJI_0001.JPG to NS2_DJI_0632.JPG
EW1_DJI_0001.JPG to EW1_DJI_0999.JPG
EW2_DJI_0001.JPG to EW2_DJI_0654.JPG
OB1_DJI_0001.JPG to OB1_DJI_0999.JPG
OB2_DJI_0001.JPG to OB2_DJI_0566.JPG
Strangely, loading the images in alphabetical order is the way that fails. Loading them in non-alphabetical order seems to succeed.

Johnny555555, 2022-06-07:
I've been comparing log files when processing fails after about 45 minutes and I notice some commonality:
Example a)
File "/usr/local/lib/python3.8/dist-packages/numpy/lib/npyio.py", line 112, in zipfile_factory
return zipfile.ZipFile(file, *args, **kwargs)
File "/usr/lib/python3.8/zipfile.py", line 1269, in __init__
self._RealGetContents()
File "/usr/lib/python3.8/zipfile.py", line 1336, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
terminate called without an active exception
/code/SuperBuild/install/bin/opensfm/bin/opensfm: line 12: 175150 Aborted                 (core dumped) "$PYTHON" "$DIR"/opensfm_main.py "$@"

Example b)
_zip = zipfile_factory(fid)
File "/usr/local/lib/python3.8/dist-packages/numpy/lib/npyio.py", line 112, in zipfile_factory
return zipfile.ZipFile(file, *args, **kwargs)
File "/usr/lib/python3.8/zipfile.py", line 1269, in __init__
2022-05-25 09:15:20,626 DEBUG: Matching EW2_DJI_0569.JPG and NS2_DJI_0259.JPG.  Matcher: FLANN (symmetric) T-desc: 2.837 T-robust: 0.016 T-total: 2.860 Matches: 168 Robust: 139 Success: True
self._RealGetContents()
File "/usr/lib/python3.8/zipfile.py", line 1336, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
terminate called without an active exception
/code/SuperBuild/install/bin/opensfm/bin/opensfm: line 12:  4804 Aborted                 (core dumped) "$PYTHON" "$DIR"/opensfm_main.py "$@"

Johnny555555, 2022-06-08:
Example c)
_zip = zipfile_factory(fid)
File "/usr/local/lib/python3.8/dist-packages/numpy/lib/npyio.py", line 112, in zipfile_factory
return zipfile.ZipFile(file, *args, **kwargs)
File "/usr/lib/python3.8/zipfile.py", line 1269, in __init__
2022-06-08 00:44:58,418 DEBUG: Matching OB2_DJI_0023.JPG and OB1_DJI_0265.JPG.  Matcher: FLANN (symmetric) T-desc: 2.551 T-robust: 0.028 T-total: 2.579 Matches: 37 Robust: 10 Success: False
self._RealGetContents()
File "/usr/lib/python3.8/zipfile.py", line 1336, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
terminate called without an active exception
terminate called recursively
/code/SuperBuild/install/bin/opensfm/bin/opensfm: line 12: 481253 Aborted                 (core dumped) "$PYTHON" "$DIR"/opensfm_main.py "$@"

Johnny555555, 2022-06-09:
I've been testing with different loading sequences and varying volumes of images and have now replicated the same error with a different file sequence & quantity.
I think what's happening is there is an underlying issue which causes corrupt zip files to be created during the feature matching in the SFM processing stage. It is exacerbated by higher resolution images, larger quantities of images, and the sequence of loading images seems to have an impact also.

This is the common error that is displayed in the logs:
File "/usr/lib/python3.8/zipfile.py", line 1336, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
terminate called without an active exception
terminate called recursively
/code/SuperBuild/install/bin/opensfm/bin/opensfm: line 12

Johnny555555, 2022-06-09:
I repeated a test 3 times with 2400 images selected in the same order (NS1,EW1,OB1,NS2,EW2,OB2). It failed the first two times (with the bad zip file error) and completed successfully the third time.

Johnny555555, 2022-08-14:
To close this out - the root cause was the race condition when using a GPU. It was fixed in the nodeodm:gpu version released on about 1st August 2022.

closed 2022-08-14 by Johnny555555