WebODM/WebODM #1189: Clarify token expiration time settings

cheoppy, 2022-05-28:
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 100 on Windows 10

### What is the problem?

The Token Expiration section of the WebODM document (https://docs.webodm.org/#token-expiration) states that "The expiration time is dependent on WebODM’s settings.", which may imply that token expiration time can be configured in the WebODM settings. However it is defined to be 6 hours here:
https://github.com/OpenDroneMap/WebODM/blob/master/webodm/settings.py#L330 which, if we want to change, requires tackling with the Django settings within the WebODM docker image. I would either change the docs to state that the default token expiration time is 6 hours or make the token expiration time setting available through environment variables.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Running a dataset processing that takes more than 6 hours.

pierotofy, 2022-05-29:
Hey @cheoppy :hand: thanks for the report.

We could indeed improve the documentation.

Would you be interested in opening a pull request to improve the wording?

cheoppy, 2022-05-29:
Hi @pierotofy ,

Sure! I looked up how the documentation is being generated, and I found that this may be the place to start:
https://github.com/OpenDroneMap/WebODM/blob/master/slate/source/includes/reference/_authentication.md

It is the file that I may be looking for?

pierotofy, 2022-05-30:
Yes.

closed 2022-07-07 by pierotofy