WebODM/WebODM #1191: Possible Bug: Broken app_logo and favicon when using --media-dir redirect from Docker storage

Saijin-Naib, 2022-05-31:
### How did you install WebODM (docker, installer, etc.)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Alpine Linux Edge (v3.16) with Microsoft Edge

### What is the problem?
After setting the --media-dir flag, WebODM has a broken/missing app_logo and favicon. This behavior is consistent on two machines tested locally with Microsoft Edge, Mozilla Firefox, and GNOME Web (Epiphany).
```
webapp      | WARNING Cannot get app_logo_favicon, this could mean the image was deleted.
webapp      | WARNING Cannot get app_logo_36, this could mean the image was deleted.
```
### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
Launch webodm.sh start --media-dir with a local directory, observe visually missing iconography on most page loads/transitions:  
[image]

Console log warnings above.

Saijin-Naib, 2022-07-09:
Seemingly a ```./webodm.sh rebuild``` resolved the missing icons, if that helps any.

Sjullep, 2022-10-26:
Yes, this helped, thanks

closed 2026-11-04 by pierotofy