WebODM/WebODM #1214: Feature Request: libcaesium integration

Saijin-Naib, 2022-07-26:
### How did you install OpenDroneMap? (Docker, natively, ...)?
Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
NA

### What is the problem?
JPEGs coming out of most sensors are not well-optimized, and there are a host of lossless optimizations that can be applied to them to reduce file size. Typically, this is on the order of 20% or so.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
I think looking at libcaesium for ingested images into ODM would be neat, as it could result in a smaller local database store, smaller upload to the cloud for Lightning, and smaller shuffling around for cluster/node operations.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Run Caesium Image Compressor against a dataset.
SOOC
[image]

Lossless Optimization with libcaesium
[image]

pierotofy, 2024-05-31:
Not enough bandwidth for this.

closed 2024-05-31 by pierotofy

Saijin-Naib, 2024-05-31:
It looks like libcaesium needs some work upstream to keep metadata when transcoding formats, so I will open this again later when that is addressed