WebODM/WebODM #1216: OpenMVS failed with GPU.

gnotarides, 2022-07-28:
Hi all,
On an HP Zbook 15G2 with a Quadro K2100M (2 GB Vram), running on W 10 64bit I get the 
"OpenMVS failed with GPU, is your graphics card driver up to date? Falling back to CPU." message (while running a 3d Model task of a 98 drone photos with 7 GCPs project).
The installation is via the installer not via docker. 
The driver is not that old from what I can understand (426.52 installed 426.78 latest).
Is there something I can do or is it just a low VRAM card, ignored by WebODM?
Thank you.

pierotofy, 2022-07-28:
Hi @gnotarides,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-07-29 09:15) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2022-07-28:
According to https://developer.nvidia.com/cuda-legacy-gpus the 2100 series has CUDA compute capability 2.1. You need at least 3.5.

There's unfortunately nothing we can do, NVIDIA has pulled the ability to compile for 2.x compute from its compiler.

closed 2022-07-28 by pierotofy