WebODM/WebODM #1217: Custom boundary file with extension ```.geojson``` cannot be selected in the relevant dialogue, requires renaming to ```.json```

ivangayton, 2022-07-31:
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 103 on Linux

### What is the problem?

When uploading a GeoJSON file as a custom boundary, the Web interface only displays/permits files with extension .json, not .geojson. 

Workaround: rename the GeoJSON file to have a .json extension. 

Desired behaviour: valid GeoJSON files with extension ```.json``` or ```.geojson``` are selectable, eliminating the necessity to rename GeoJSON files for use as boundaries.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

As far as I can see, any installation of WebODM will do this. I think it's probably just a filter in the dialogue that filters out anything except files with a ```.json``` extension (which makes sense for many of the other parameters such as the cameras file) and simply doesn't "know" that ```.geojson``` is also a valid JSON extension and—if properly formed—will contain valid JSON syntax.

Specifically:

- Start any processing project
- Choose ```Options -> Edit``` for the parameters
- Beside the ```boundary (json)``` box, click the ```Click to import a json file``` button
- Navigate to a directory containing a valid GeoJSON file
- Note that it is not visible in the dialogue, and therefore cannot be selected for upload
- In the file explorer, rename the file with extension ```.json```
- Back in the WebODM dialoge, the file appears and can be selected; on selection the contents appear in the parameter string and seem to work fine.

Saijin-Naib, 2022-07-31:
Have you observed this behavior with Firefox or Epiphany on Linux as well?

ivangayton, 2022-07-31:
Yes, with Firefox.

However, I've just noticed/realized that there's a second workaround here: you can remove the filetype filter from the dialogue in Firefox.  And once I realized that, I saw that it can be done in Chrome as well.

The default for the upload dialogue in Firefox is a filter for ```All Supported Types``` (others include ```Javascript Object Notation```, ```Text File```, and ```All Files```. If you set it to All Files you can grab the GeoJSON without renaming. 

In Chrome there are only two options, ```Custom Files``` and ```All Files```. Set to ```All Files``` you can grab the GeoJSON directly. 

Apologies for not noticing this beforehand; it was already a pretty minor problem and I now realize it's even more minor. That being said, I still think it makes sense for files with a ```.geojson``` extension to be visible in the ```All Supported Types``` filter view (Firefox) or ```Custom Files``` view (apparently the equivalent in Chrome). When you choose ```All Files``` you may have to wade through a lot of files that are neither JSON nor GeoJSON, which is not a big deal but not the optimal experience.

So the appropriate description of the desired behaviour is: valid GeoJSON files with extension .json or .geojson are selectable _with the default browser dialogue settings (```All Supported Types``` in Firefox or ```Custom Files``` in Chrome)_, eliminating the necessity to _either change the filter in the dialogue (and therefore see all files in the directory, most of which are probably not the GeoJSON boundary file you are trying to upload) or_ rename GeoJSON files for use as boundaries.

Saijin-Naib, 2022-07-31:
I've fixed the file filter for another file, so I'll work on the same for this. It should just be transparent when I have it done right.

pierotofy, 2022-10-19:
Fixed by https://github.com/OpenDroneMap/WebODM/commit/7ab2b14ba60cf5eed5ee42257d0fc303f7bc4182 thanks for the report!

closed 2022-10-19 by pierotofy

ivangayton, 2022-10-19:
Thanks! Great work!