WebODM/WebODM #1218: After running ./webodm.sh, the process seems to be looping

Eri2726, 2022-07-31:
I've tried the following code.
`./webodm.sh start`

But this process has been going on for about two hours.
Is this the correct behavior?

The following picture is on the terminal screen that is on going process seems to be looping.

[image]

pierotofy, 2022-07-31:
Hi @Eri2726,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-08-01 14:35) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Eri2726, 2022-07-31:
### How did you install WebODM (docker, installer, etc.)?

this is a docker install on windows, but the problem also appears with docker.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 103 on Windows 8.1

### What is the problem?

I've tried the following code.
`./webodm.sh start`

But this process has been going on for about two hours.
Is this the correct behavior?

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Run  the folloeing code from the Docker terminal.
.`/webodm.sh start`

and the image below is the terminal screen that it has been processing.
[image]

Eri2726, 2022-07-31:
I can resolve that!
Thank you.

closed 2022-07-31 by Eri2726