WebODM/WebODM #1219: WebOdm docker containers uses a huge amount of memory

diegoaces, 2022-08-05:
### How did you install WebODM (docker, installer, etc.)?

Docker 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

### What is the problem?

Docker webodm container uses a huge amount of memory 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

When updated the webodm system with ./webodm.sh update  command  there is garbage left over from the old containers.

[image]

[image]

I know that running the docker image prune -a command frees up memory, but is there a way to automatically decrease the amount of space that docker uses?

pierotofy, 2022-08-05:
Mm, not really; I mean, we could automatically run `docker image prune` for the user, but I'm afraid this is not what some end users would like, as WebODM might not be the only application that runs on docker.

diegoaces, 2022-08-05:
Thanks @pierotofy, you are right, for those who have other containers running in docker it would cause problems.
But it is a point to consider.

closed 2022-08-05 by diegoaces