WebODM/WebODM #1226: Custom EPSG

dcruse67, 2022-08-25:
### How did you install WebODM (docker, installer, etc.)?

Installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Brave Browser & Windows 10 Pro

### What is the problem?

PROJ: Cannot open https://cdn.proj.org/us_noaa_nchpgn.tif: Network functionality not available

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Export any Ortho, Surface, or Plant Health Image using Custom EPSG in the Projection settings (EPSG 2240 selected for my area which is State Plane Grid - GA West). This functionally needed to export maps to a useful scale when being imported as an overlay into an AutoCAD files that are on the US State Plane Grid System. Other settings that do work are badly out of scale (too small) and not anywhere close to the correct datum.

SabraHashemi, 2022-11-23:
I have the same issue, 
I tested your introduced datasets and I got an orthomosaic image that is a little smaller than the map and its location is not very accurate.

rstoresund, 2023-07-13:
Was there a resoultion to this issue?  I"m having this problem as well (2023-07-12).
[image]

pierotofy, 2025-07-17:
This is a Windows native specific issue @smathermather, you might want to look into it to track it / fix it, I will close this one if there's no activity because I don't have anything to fix inside WebODM's codebase. 🙏

pierotofy, 2025-07-17:
Yes that's OK, but it would be nice to see progress being made toward resolving these, what's the plan?

Saijin-Naib, 2025-07-17:
We are going to discuss these and other issues with @Sylveon to see if we can't fix them up.

smathermather, 2025-07-17:
Takes a bit of time to recruit and onboard. Go slowly before going quickly and build great and sustainable teams. More to come.

Also, as we have a tradition of closing issues before they are complete on github, do we need to set up a separate issue tracker so we don't lose sight of ones like this if they are closed before addressed if there's no champion for them at that time?

Happy to set something up concurrent with github, just a temperature check on how to manage github issues.