WebODM/WebODM #1236: Unable to view thermal imagery in multispectral project

duckblaster, 2022-09-09:
Something wrong in /app/api/formulas.py, whenever I select thermal from algorithm drop down it fails with the error below:

Cannot find band "Lwir" from "BGRNLwir". Choose a proper band order.

### How did you install WebODM (docker, installer, etc.)?

Docker WebODM 1.9.15

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 105 on Windows 11]

### What is the problem?

Whenever I select thermal from algorithm drop down it fails: 
```
Cannot find band "Lwir" from "BGRNLwir". Choose a proper band order.
```

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Try to view/download the thermal view of a multispectral orthoimage.

pierotofy, 2022-09-09:
Hi :wave: thanks for the report; can you share your RAW orthophoto GeoTIFF?

duckblaster, 2022-09-11:
Here's the whole project output: https://1drv.ms/u/s!AsY5aB0MpC2Dgd5vRRAJiF6zPtdxew?e=FJ8B6n

pierotofy, 2022-09-12:
The logic to parse the LWIR band was incorrect;

This has now been fixed.

[image]

closed 2022-09-12 by pierotofy

duckblaster, 2022-09-16:
Thanks, that fixed it!
Some of the functions are not setting the correct min and max values automatically though:
[image]

pierotofy, 2022-10-01:
This should be improved with #1246

PR #1238 (2022-09-12): Fix band order bug, temperature formulas
Closes #1236﻿