WebODM/WebODM #1243: Installation guide is not correct for Windows

turkeyphant, 2022-09-26:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************
### How did you install WebODM (docker, installer, etc.)?

Following https://docs.opendronemap.org/installation/

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Chrome 105 on Windows 10](https://www.whatismybrowser.com/detect/what-version-of-chrome-do-i-have)

### What is the problem?

Following installation guide at  https://docs.opendronemap.org/installation/

1. Turned on virtualisation in BIOS.
2. Installed Python 3
3. Installed Docker for Windows
4. cloned https://github.com/OpenDroneMap/WebODM repo
5. Launched WebODM (instructions are incorrect as they don't say to navigate to repo first)
6. `webodm.sh` stops here:

```
broker             | 1:M 26 Sep 2022 22:27:25.780 * Running mode=standalone, port=6379.
broker             | 1:M 26 Sep 2022 22:27:25.780 # Server initialized
broker             | 1:M 26 Sep 2022 22:27:25.780 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
broker             | 1:M 26 Sep 2022 22:27:25.781 * Loading RDB produced by version 7.0.5
broker             | 1:M 26 Sep 2022 22:27:25.781 * RDB age 15 seconds
broker             | 1:M 26 Sep 2022 22:27:25.781 * RDB memory usage when created 0.82 Mb
broker             | 1:M 26 Sep 2022 22:27:25.781 * Done loading RDB, keys loaded: 0, keys expired: 0.
broker             | 1:M 26 Sep 2022 22:27:25.781 * DB loaded from disk: 0.000 seconds
broker             | 1:M 26 Sep 2022 22:27:25.781 * Ready to accept connections
db                 |
db                 | PostgreSQL Database directory appears to contain a database; Skipping initialization
db                 |
db                 | LOG:  database system was shut down at 2022-09-26 22:27:10 UTC
db                 | LOG:  MultiXact member wraparound protections are now enabled
db                 | LOG:  database system is ready to accept connections
db                 | LOG:  autovacuum launcher started
webodm-node-odm-1  | info: Authentication using NoTokenRequired
webodm-node-odm-1  | info: Listening on 0.0.0.0:6367 UDP for progress updates
webodm-node-odm-1  | info: No tasks dump found
webodm-node-odm-1  | info: Checking for orphaned directories to be removed...
webodm-node-odm-1  | info: Server has started on port 3000
Error response from daemon: Ports are not available: exposing port TCP 0.0.0.0:8000 -> 0.0.0.0:0: listen tcp 0.0.0.0:8000: bind: An attempt was made to access a socket in a way forbidden by its access permissions.
```

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

As above. Just run `webodm.sh` on Windows.

pierotofy, 2022-09-27:
Is there some other service occupying port 8000? Try to launch using: `./webodm.sh restart --port 10000`

Anyway, this doesn't look like a software fault in WebODM per-se (if it's an error in the docs, this should be reported at https://github.com/OpenDroneMap/docs/issues ) :pray:

closed 2022-09-27 by pierotofy

turkeyphant, 2022-09-27:
Dunno what could be using that port but thanks, that fixed it :)