WebODM/WebODM #1245: Unable to download Assets, large files over 300 MB

implan-sig, 2022-09-30:
### How did you install WebODM (docker, installer, etc.)?

Docker on Ubuntu Linux 22.04 GCP VM

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Edge 105 on Windows 11](https://www.whatismybrowser.com/detect/what-version-of-edge-do-i-have)

### What is the problem?

After the processing finishes I have tried to download the result files from multiple computers and internet connections, but the download always fails if the all.zip is over 300 MB approximately, the download gives a Couldn't download - Network issue error in under a minute, only some smaller files will be downloaded.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

(https://webodm.implantijuana.gob.mx/public/task/ac10732d-f1fd-4c33-9853-8f388588aeb2/map/)
[image]

implan-sig, 2022-09-30:
I'm guessing that the GCP Load Balancer is causing the network connection to be interrupted. I tried downloading using the IP address instead of the domain name and it works at a much low download speed.

pierotofy, 2022-10-01:
If this is a configuration issue with the GCP Load Balancer then this can probably be closed.