WebODM/WebODM #1248: Max images count exceeded error message

diegoaces, 2022-10-07:
### How did you install WebODM (docker, installer, etc.)?

Docker and mounted on 200 gb volume to storage the processing output.  
WebODM Lightning Network - Unlimited.

### What's your browser and operating system?

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
Ram:   8 GB

### What is the problem?

```
[2022-10-07 13:24:30,616: INFO/ForkPoolWorker-2] worker.tasks.process_task[xxx-xxx-xxx]: Processing... Task [Constitución - 5/10/2022] (xxx)
webapp    | INFO Booting WebODM 1.9.15
[2022-10-07 13:42:10,983: ERROR/ForkPoolWorker-2] worker.tasks.process_task[xxx-xxx-xxx]: FAILURE FOR Task [xxx - 5/10/2022] (xxx): Max images count exceeded.
```

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Created at : 6/10/2022, 22:46:58
Node:  Nodo 1 (manual)
Options:  auto-boundary: true, dsm: true

[image]

This problem does not occur for processes with less than 500 images.

pierotofy, 2022-10-07:
Hey @diegoaces ✋ did you make sure your account has sufficient credits or an unlimited plan? Triple check that your node token matches that of your webodm.net account also.

diegoaces, 2022-10-07:
Hi @pierotofy, yes I have an unlimited plan, in the image you can see another process carried out almost simultaneously with the one that causes problems. Also mention that there are two processes that I can perform in parallel.

[image]

pierotofy, 2022-10-07:
Is your internet connection stable/reliable? @Saijin-Naib have you see this one before?

diegoaces, 2022-10-07:
It is an ODM server running on a DigitaOcean droplet.
After many attempts it was able to process correctly.

closed 2022-10-07 by diegoaces

Saijin-Naib, 2022-10-08:
@pierotofy , yeah, once or twice with really flaky uplinks, usually from clients who do not have a DigitalOcean server in their geographic region. I think parts of the upload timeout or drop and the reconnecting messes with the image count somehow... Or that's what it feels like from the few times I saw it.

Similar to @diegoaces , one customer resolved it by just repeatedly re-trying, and another resolved it by traveling to a larger city with better connectivity.

pierotofy, 2022-10-08:
Mm, I wonder if https://github.com/OpenDroneMap/ClusterODM/blob/c2c2e8f97b9518055f97e822cd5e5874dbc9be52/libs/taskNew.js#L170 is not being called in certain instances (e.g. connection timeout). I've opened https://github.com/OpenDroneMap/ClusterODM/issues/99