WebODM/WebODM #1250: Cannot process dataset of 334 images taken at 45° angle

implan-sig, 2022-10-14:
### How did you install WebODM (docker, installer, etc.)?

Docker on Ubuntu Linux 22.04 GCP VM

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
https://whatismybrowser.com/w/GS54EAW
[Edge 106 on Windows 10]

### What is the problem?

I uploaded the files and set the high quality preset, after some time of processing I get the Cannot process dataset error with the follwing information:

21:57:28 [App     ] Build date: Sep 20 2022, 13:07:58
21:57:28 [App     ] CPU: Intel(R) Xeon(R) CPU @ 2.20GHz (2 cores)
21:57:28 [App     ] RAM: 15.62GB Physical Memory 0B Virtual Memory
21:57:28 [App     ] OS: Linux 5.15.0-1019-gcp (x86_64)
21:57:28 [App     ] Disk:
21:57:28 [App     ] SSE & AVX compatible CPU & OS detected
21:57:28 [App     ] Command line: DensifyPointCloud /var/www/data/1e409a4e-6ce4-43ca-8e59-f46116f25e29/opensfm/undistorted/openmvs/scene.mvs --resolution-level 1 --dense-config-file /var/www/data/1e409a4e-6ce4-43ca-8e59-f46116f25e29/opensfm/undistorted/openmvs/Densify.ini --min-resolution 1024 --max-resolution 2048 --max-threads 2 --number-views-fuse 2 --sub-resolution-levels 2 -w /var/www/data/1e409a4e-6ce4-43ca-8e59-f46116f25e29/opensfm/undistorted/openmvs/depthmaps -v 0 --cuda-device -2 --geometric-iters 0
21:57:28 [App     ] Find a camera not pointing towards the scene center; the scene will be considered unbounded (no ROI)
21:57:39 [App     ] Preparing images for dense reconstruction completed: 334 images (10s532ms)
21:57:42 [App     ] Selecting images for dense reconstruction completed: 0 images (3s292ms)

Fused depth-maps 0 (100%, 0ms)
21:57:42 [App     ] Densifying point-cloud completed: 0 points (13s847ms)
21:57:42 [App     ] MEMORYINFO: {
21:57:42 [App     ] 	VmPeak:	 1159976 kB
21:57:42 [App     ] 	VmSize:	 1087712 kB
21:57:42 [App     ] } ENDINFO
[INFO]    running "/code/SuperBuild/install/bin/OpenMVS/DensifyPointCloud" --filter-point-cloud -1 -i "/var/www/data/1e409a4e-6ce4-43ca-8e59-f46116f25e29/opensfm/undistorted/openmvs/scene_dense.mvs" -v 0 --cuda-device -2 --geometric-iters 0
21:57:42 [App     ] Build date: Sep 20 2022, 13:07:58
21:57:42 [App     ] CPU: Intel(R) Xeon(R) CPU @ 2.20GHz (2 cores)
21:57:42 [App     ] RAM: 15.62GB Physical Memory 0B Virtual Memory
21:57:42 [App     ] OS: Linux 5.15.0-1019-gcp (x86_64)
21:57:42 [App     ] Disk:
21:57:42 [App     ] SSE & AVX compatible CPU & OS detected
21:57:42 [App     ] Command line: DensifyPointCloud --filter-point-cloud -1 -i /var/www/data/1e409a4e-6ce4-43ca-8e59-f46116f25e29/opensfm/undistorted/openmvs/scene_dense.mvs -v 0 --cuda-device -2 --geometric-iters 0
21:57:42 [App     ] error: no valid point-cloud for the ROI estimation

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "/code/stages/odm_app.py", line 88, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 378, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 378, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 378, in run
self.next_stage.run(outputs)
[Previous line repeated 1 more time]
File "/code/opendm/types.py", line 359, in run
self.process(self.args, outputs)
File "/code/stages/openmvs.py", line 206, in process
system.run('"%s" %s' % (context.omvs_densify_path, ' '.join(config + gpu_config + extra_config)))
File "/code/opendm/system.py", line 109, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
I used the following preset:
[image]
The error I get:
[image]
The dataset including the console.txt file:
https://1drv.ms/u/s!AsZI_CsMoVvfj4djqaDi5uTtoxoNWQ?e=rvLNp2

pierotofy, 2022-10-15:
Thanks for sharing the dataset! 

Did you use a mission planner to capture the images? Do they have sufficient overlap? You should have at least 70% for all images.

implan-sig, 2022-10-16:
> Thanks for sharing the dataset!
> 
> Did you use a mission planner to capture the images? Do they have sufficient overlap? You should have at least 70% for all images.

Thanks for looking into this issue, this is the 3rd time it happens, with 3 different datasets, the other datasets where taken at different times of the day, or on multiple days, I figured that was the problem, but this one was taken in a single flight.

I used Map Pilot Pro from Maps Made Easy, it was taken with at a height of 45 m and 85% overlap. Here, are the settings:

[image]

pierotofy, 2022-10-17:
Mm, strangely this reconstructed on my Windows machine (looks nice too!):

[image]

I'm running a test on webodm.net which uses the latest version and is on Linux, we'll see if it processes there.

Note that I didn't resize the images prior to processing (WebODM will do that by default). Have you tried without resizing?

implan-sig, 2022-10-17:
Wonderful, I'll give it a try without resizing and send my results.

Thank you

pierotofy, 2022-10-17:
Ok this processed on a docker machine without resizing also: https://wln.s3.webodm.net/432c8dad-46a0-41f5-a3bb-5cf085f15b25/all.zip

implan-sig, 2022-10-18:
It took some time but it was able to process by turning off resizing, as you said.

[image]

My result crops the point cloud, how do you keep the all the data as it is shown in your picture?

Thank you

pierotofy, 2022-10-18:
Turn off `--auto-boundary` if you've selected it.

closed 2022-10-18 by pierotofy